ArduPilot/ardupilot #13092: Matek F765 and TriTilt VTOL: Yaw Servo Parameters MID/MIN/MAX are reset when booting

Rolf-G, 2019-12-22:
### Bug report
**Issue details**
In Tri-Tilt VTOLs with continous YAW-Servo (Q_TILT_TYPE=0) the ServoX_function=39 (usually motor 7) controls the YAW servo. But ServoX_MIN/MAX/MID Values adapted to the servo are always reseted when booting to MID 1000, MIN 1000, MAX 2000 (As if there is a motor and not a servo).

I noticed this behavior when replacing a Pixhawk with an F765 Wing and installing ArduPlane V4.0.2beta2 (46b36493). 

The error with identical firmware does NOT occur on a Pixhawk1-1M.

**Version**
ArduPlane V4.0.2beta2 (46b36493). 

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Q_FRAME_CLASS,7  (Tri)
Q_FRAME_TYPE,6  (reversed Tri)
Q_TILT_TYPE,0 

**Hardware type**
MatekF765-Win 00260038 30385102 3935333

**Logs**
Still no Logs

Rolf

tridge, 2019-12-24:
thanks, I've reproduced the issue. It is a bit late at night for me to work on it now, but I should have some time in the next couple of days

tridge, 2019-12-24:
this is actually deliberate. I've opened a wiki issue to add some docs on it:
https://github.com/ArduPilot/ardupilot_wiki/issues/2358

closed 2019-12-24 by tridge

Rolf-G, 2019-12-26:
I found out, that it  works with a better parameter setting! Sorry for the irritation that I caused. 
Instead of SERVO_BLH_AUTO 1 (set it to 0) set SERVO_BLH_MASK with the output-bitmask of the actually connected BLHELI-ESCs. Telemetry in non-vector tri-motor VTOLs will work.