ArduPilot/ardupilot #13093: Multiple PID Profiles

CodingGhost, 2019-12-22:
### Feature request
I would like a feature where you can add multiple PID Profiles to the config and select them with an AUX channel.

I need this, because I use my plane with different setups during one session (different battery weights + Gopro/no Gopro)

an alternative to using AUX channels could be to select it pre flight with some sort of stick input or with missionplanner. (but I think this will be harder to implement)

I think it shouldnt be too hard to implement, and if somebody points me in the right direction where I have to modify/insert some code, I would be happy to help.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

Pedals2Paddles, 2019-12-22:
I think you'd need to double the parameters for each one first. Then you'd need to modify every piece of code that uses them with a case switch based on aux input or whatever is going to drive it.

I don't doubt there is a use case for it. But this is probably better handled by a companion computer or AP_Scripting.

CodingGhost, 2019-12-23:
couldnt you just still use the same variables and just select what values to load into these before arming? then you wouldnt have to insert switches everywhere.

but anyways, how would I use AP_Scripting for that?

proficnc, 2019-12-23:
The use cases for this are spraying drones, and delivery drones.
Where tuning may be quite different from full to empty

SergeyBokhantsev, 2019-12-23:
https://github.com/ArduPilot/ardupilot/issues/6680

Pedals2Paddles, 2019-12-23:
> couldnt you just still use the same variables and just select what values to load into these before arming? then you wouldnt have to insert switches everywhere.

Select values from where? It sounds like you're suggesting a GCS function to load a premade set of parameters from a file? This is a very basic existing GCS function already. This is by far the simplest way to do what you need. And this can be automated with companion computer too.

If you're suggesting on board storage of these other values in memory, that's again doubling the parameters like I suggested above.

CodingGhost, 2019-12-23:
yes, ofcourse onboard storage, and no, you wouldnt have to double it if you do it like described in my last post on #6680, wouldnt that be a possibility?

IamPete1, 2019-12-23:
will be possible with scripting once this goes on https://github.com/ArduPilot/ardupilot/pull/12889

Pedals2Paddles, 2019-12-23:
AP_Scripting is by far and away the method to use for this kind of thing if you don't want a companion computer.  You can look for whatever RC input you want, and load whatever parameters you want, without having to reinvent how the PID parameters are handled now.

rmackay9, 2019-12-24:
I also agree that in the short-term at least, scripting is going to be the best way to handle this so we should concentrate on trying to get @IamPete1's PR in.

CodingGhost, 2019-12-24:
ok, if #12889 gets resolved, I think this is a good solution! I will post the script here when im done

IamPete1, 2021-07-26:
The scripting methods are now in, we also have fast access so you can fiddle the PIDs in real time if you need to.