ArduPilot/ardupilot #13101: Copter: add Brake RC failsafe option

rmackay9, 2019-12-23:
This is an enhancement request to add Brake mode as an RC failsafe option for Copter.  There are probably two cases where this is a good idea:

- Indoor drones especially swarms although Land (an existing option) seems like a good solution for this
- Outdoor tethered drones where there's little danger of the vehicle getting stuck

This new option is a very bad idea for general use case because if the user can't get RC working again somehow (perhaps by moving closer to the drone or changing the transmitter's antenna position) the vehicle could get stuck until it's battery runs low and a battery failsafe kicks in.

We can imagine a really bad situation where the user has failed to setup the battery or GCSs failsafes and has chosen this new option and must then simply wait until the vehicle runs out of batteries (and likely crashes).

Because of the above reasons we should ensure that this option comes with safety warnings that it should not be used in most cases.

Pedals2Paddles, 2019-12-23:
I feel like this is an option that shouldn't even be user selectable.  Have it in the code, disabled, and a user would need to compile it on their own to use it.  Seems like someone who has a use case would probably be a power user enough to do that.  People don't read the docs and warnings.

rmackay9, 2023-06-30:
We can close this issue as complete now.  txs

closed 2023-06-30 by rmackay9