ArduPilot/ardupilot #13110: Plane: Allow Pre-Arm check pass without compasses

Naterater, 2019-12-25:
### Bug report

**Issue details**
When using plane without a compass, the INS pre-arm check will not pass... according to https://discuss.ardupilot.org/t/ahrs-not-healthy-on-arm/50239/

I recommend NOT disabling INS pre-arm checks in order to pass; we should allow passing, but maybe write a status text indicating that no compasses are enabled when arming.

**Version**
Plane 4.0

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
Presumably Matek

**Logs**
None given in the discuss topic.

kd0aij, 2019-12-26:
I think setting COMPASS_USE to zero is a reasonable way to indicate that no compass is present, and in that case, the INS prearm check should pass.

peterbarker, 2020-01-07:
@kd0aij That's current behaviour, isn't it?

So can we close this?

kd0aij, 2020-01-07:
If @Naterater agrees...

Naterater, 2021-10-04:
I guess this is still an issue: https://discuss.ardupilot.org/t/ahrs-not-healthy-on-arm/50239/8?u=naterater

closed 2022-09-11 by IamPete1