ArduPilot/ardupilot #13117: Spektrum: 22ms protocol unsupported

Naterater, 2019-12-26:
### Bug report

Many users can't get signals to ardupilot as the default "frame rate" of spektrum transmitters is 22ms. The 11ms protocol works.

https://discuss.ardupilot.org/t/spectrum-dx8-no-connection-to-pixhawk2-1/49742/

I also answered the question at least twice on facebook, and I'm surprised I stumbled on the answer myself after trying multiple spektrum transmitters to bind the SPM9645.

**Version**
Plane 4.0, probably all vehicles though

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Cube blue mini carrier board, but reported on others as well. SPM9645 satellite receiver.

**Logs**
N/A

rmackay9, 2020-01-09:
Looks like we have a fix for this issue now in this PR: https://github.com/ArduPilot/ardupilot/pull/13211

Ernestoide, 2020-02-05:
Hi.  I'm following this trying to solve the same or similar issue.  I'm using a Spectrum DX6e radio with a AR620 receiver.  After digging in the web finally i found the way to correctly wire the receiver to the DSM port in the Pixhawk.    I can power the receiver and bind it with my radio while connected to the Pixhawk but still no response from the mission planner radio calibration screen.  I'm using the plane version 4.03 that suppose to fix the frame rate issue.  

I tried using the option in mission planner for binding DSMx but it just do nothing.  

I really don;t know what else to try.   Thanks so much in advance.

andyp1per, 2020-02-05:
@Ernestoide there is another fix you need that only just got pushed this morning - so probably wait for the next rc to be released and you will be good.

Ernestoide, 2020-02-05:
Thanks  @andyP1per  is that part of 4.04 ?

andyp1per, 2020-02-05:
I don't know what Plane's release schedule is - it will go out with copter tomorrow I believe.

Ernestoide, 2020-02-05:
ok I'll give a try with the copter firmware

andyp1per, 2020-02-06:
@Ernestoide my apologies, the fix will only affect you if you are using a pixracer. If you are not then it is something else.

Ernestoide, 2020-02-06:
I'm using a Pixhawk.  any other users out there with similar issues?  Any suggestions?

I really appreciate your help

andyp1per, 2020-02-06:
Are you able to build the firmware? I think it's worth switching on DSM_DEBUG in that file and see what it tells you about your setup.

Ernestoide, 2020-02-07:
I'm new to Ardupilot  and my coding days are long gone.  I could run some debug if you can point me out to some sources on how to do it.  My other choice is to change receiver and/or radio or give up all together :(

andyp1per, 2020-02-07:
@Ernestoide if you can tell me the exact board you are using I can do you a build of plane that includes the debug message

Ernestoide, 2020-02-07:
@andyp1per   Thanks so much.

I'm using a Pixhawk  2.4.8   .  My receiver is an spectrum AR620 and my transmitter is an Spectrrum DX6e

2/7/2020 10:25:39 AM : fmuv2 0026003A 31385117 39393931
2/7/2020 10:25:39 AM : ChibiOS: cb42c858
2/7/2020 10:25:39 AM : ArduPlane V4.0.3 (46322788)

Please let me know if there is any other information needed.

Thanks
[image]

[image]

andyp1per, 2020-02-07:
I'm pretty sure you won't have the 1Mb limit, here is a build with debug on:

https://www.dropbox.com/sh/60ypsv0nt6tsj5x/AADgkprrNvFPlaI18xLqjHqea?dl=0

are you sure that receiver is compatible?

Ernestoide, 2020-02-07:
Firmware loaded with no issues but behavior is the same.  Is there any log files to check if the board is discovering the receiver?

This is one of those thing that is not clear to me. The receiver is a DSMX receiver and should be compatible but I cannot find this model in any list.

Thanks

andyp1per, 2020-02-07:
Sorry, to see the debug output you need to use [MAVProxy](http://ardupilot.github.io/MAVProxy/html/getting_started/index.html)

Andrewiski, 2020-03-02:
@andyp1per   What does the normal dsm debug out put look like?

I am working on a BeagleBone Blue running latest ArduCopter v4.0.3 .  I turn on DSM debug and compile.  My issue is in Mission Planner I do not have RC input on the Radio Calibration screens like I did in Arducopter 3.6.  I believe I am using 11ms not 22ms.  Setup is DX 8e to a SPM4648 receiver.

Not sure if the bug is in the DSM code or in Mission Planner 1.3.70 Build 1.3.7277.34800

DSM Debug output looks like this.

dsm state: DSM_DECODE_STATE_SYNC, count: 2, val: ff

dsm state: DSM_DECODE_STATE_DESYNC, count: 0, val: 78

dsm state: DSM_DECODE_STATE_SYNC, count: 1, val: ff

dsm state: DSM_DECODE_STATE_SYNC, count: 2, val: 61

dsm state: DSM_DECODE_STATE_DESYNC, count: 0, val: fc

dsm state: DSM_DECODE_STATE_DESYNC, count: 0, val: d8

dsm state: DSM_DECODE_STATE_SYNC, count: 1, val: 61

dsm state: DSM_DECODE_STATE_DESYNC, count: 0, val: ff

dsm state: DSM_DECODE_STATE_DESYNC, count: 0, val: 61

dsm state: DSM_DECODE_STATE_SYNC, count: 1, val: 61

dsm state: DSM_DECODE_STATE_DESYNC, count: 0, val: ff

andyp1per, 2020-03-02:
So this looks like the Pixracer bug, not sure if it's related. I suppose @tridge might be able to comment - I know the DESYNC's are bad, I don't know what you do if you get them.

andyp1per, 2020-03-02:
The Pixracer bug was that exact RX would not work with Pixracer, this was due to a DMA change in the Pixracer setup. I don't know whether Beaglebone blue would be similar as that is Pi-based I think?

Andrewiski, 2020-03-02:
It is Linux based but on a Texas Instrument arm processor so different then a Raspberry PI hardware wise.

Andrewiski, 2020-03-02:
Should I open a new issue for this since I believe I am on 11ms DSM timing?

andyp1per, 2020-03-02:
Yes probably a good idea

Andrewiski, 2020-03-02:
I opened a new issue for my report as I am at 11ms DSM timing.

[#13711](https://github.com/ArduPilot/ardupilot/issues/13711)

andyp1per, 2020-04-30:
@Ernestoide it turns out the AR620 only supports SRXL2 or PWM. With the latest firmware for the RX and this https://github.com/ArduPilot/ardupilot/pull/13923 it might now be fine.

tompano33, 2020-05-12:
Guys, may I ask which spektrum receiver works now fine with this pixhawk? Any recommendation?

andyp1per, 2020-05-12:
An extra patch was needed to make the AR620 work with SRXL2: https://github.com/ArduPilot/ardupilot/pull/14341

If you want to guarantee operation then use a Spektrum remote receiver like SPM9745 - they will all work with the existing stable code. If you want a receiver that supports telemetry (which is very nice) then get an SRXL2 receiver and use the master branch. A nice full range SRXL2 receiver is the SPM4651T. I am using SPM4650.

Naterater, 2021-07-26:
I am assuming this works with master considering the PR that was merged. If not, please reopen.

closed 2021-07-26 by Naterater