ArduPilot/ardupilot #13125: AC_Fence.cpp "get_breach_distance" function does not account for polygon fence breach

rishabsingh3003, 2019-12-28:
#### Issue details
AC_Fence.cpp (Fence Library) contains a [get_breach_distance()](https://github.com/ArduPilot/ardupilot/blob/5d70af3f76aa803f4e4a9b7675c237cefe74c26c/libraries/AC_Fence/AC_Fence.cpp#L404) function which returns the distance a vehicle has breached a fence. It is used [here](https://github.com/ArduPilot/ardupilot/blob/5d70af3f76aa803f4e4a9b7675c237cefe74c26c/ArduCopter/fence.cpp#L38) . The function, however, **does not** include **polygon fence** breaches and only checks for altitude and circle type fences. Is this done intentionally for a reason that I could not see?

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rishabsingh3003, 2019-12-28:
If someone can indeed confirm this is a bug, I would like to try and submit a patch!

peterbarker, 2019-12-29:
On Sat, 28 Dec 2019, Rishabh Singh wrote:

> If someone can indeed confirm this is a bug, I would like to try and submit a patch!

Looks like a problem to me.

I'd love to see patches and tests that fail before and pass after.

rishabsingh3003, 2020-01-01:
@peterbarker I made a small video demonstrating the issue.  I changed this[ param](https://github.com/ArduPilot/ardupilot/blob/61d5a94c45e7ae82f73089558ec503edf02c1a63/libraries/AC_Fence/AC_Fence.h#L32) to 1 meter so that its easy to recreate the issue. After an altitude/ circle fence breach occurs, another backup fence which is 20 meters ahead of the previous one is made. And if a breach occurs more than 100 meters away from the fence, a land is forced. But for polygon fence, only a one time attempt at user-specified action is taken, then no other action is imposed. 
Video: https://youtu.be/Ucp3cFpP4Eg

I have already written a function that calculates the minimum distance from inclusion/exclusion circle/polygon and calculates the distance from the breach but I am having a difficult time setting up a backup fence as done in [circle](https://github.com/ArduPilot/ardupilot/blob/61d5a94c45e7ae82f73089558ec503edf02c1a63/libraries/AC_Fence/AC_Fence.cpp#L290)/[altitude](https://github.com/ArduPilot/ardupilot/blob/61d5a94c45e7ae82f73089558ec503edf02c1a63/libraries/AC_Fence/AC_Fence.cpp#L220) fence so that when the vehicle freshly breaches a fence, a backup fence is created. And when the backup fence is breached too, the vehicle takes the appropriate action.
To make this kind of response similar for polygon fence, I feel the simplest approach would be to force one of the breach actions every 20 meters from the fence, till 100 meters is reached, where the vehicle is forced to land ( like the default setup), instead of making a backup fence. Is this a valid solution that can be merged to master or something else is required?