ArduPilot/ardupilot #13126: Tricopter yaw servo mistakenly assigned as BLHELI using SERVO_BLH_AUTO

Hwurzburg, 2019-12-29:
QPlane and presumably Copter:
in tricopter frame class, motor7 (function 39) should not be identified as BLHeli passthrough target when SERVO_BLH_AUTO=1. This resets all SERVO params each reboot and prevents proper use of yaw servo. Should be eliminated from mask if tricopter.

**Platform**
[  ] All
[  ] AntennaTracker
[x  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

Rolf-G, 2019-12-30:
Hi Henry, sorry, but it is important for understanding and for all who want to cross-check this issue:
The tricopter frame alone is not sufficient to cause the problem  as long as vectored yaw is used (because no function 39 ist used). The problem can only occur with nonvectored tricopter because only these frame use "motor 7" to control

rmackay9, 2020-01-01:
It might be best if we don't use SERVO_FUNCTION 39 (aka motor7) but instead use k_rudder or k_heli_tail_rsc.  Anyway, i think @bnsgeyer probably needs to be consulted.

IamPete1, 2021-08-15:
solved by https://github.com/ArduPilot/ardupilot/pull/18181

closed 2021-08-15 by IamPete1