ArduPilot/ardupilot #13127: Copter 4.0.0 Cube Orange SSD1306 not working

zenuavsolutions, 2019-12-29:
#### Issue details
on Cube Orange with the new 4.0.0 Copter release, the SSD1306 does not work

#### Version
Copter 4.0.0

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quad

#### Hardware type
Cube Orange

#### Logs
relevant?
[image]

RJBeckwith, 2019-12-30:
I have the same issue with my Cube Orange. Have tried two different SSD1306 displays from the recommended vendors on the AC Wiki. I get the same "snow’ on the screen. It has not worked on any of the Dev or RC versions I have tried, so didn’t expect it to work on 4.0.0.
[image]

peterbarker, 2019-12-30:
This is fixed in master.

rmackay9, 2020-01-01:
This will be released next week as part of Copter-4.0.1-rc1

closed 2020-01-01 by rmackay9