ArduPilot/ardupilot #13131: Add support for mavlink camera commands

DonLakeFlyer, 2019-12-30:
Specifically: MAV_CMD_IMAGE_START_CAPTURE, MAV_CMD_VIDEO_START_CAPTURE and MAV_CMD_SET_CAMERA_MODE. These commands were added to mavlink spec ages ago. I thought I had added an issue somewhere about them but I guess not since I can't find it. QGC uses these in it Waypoint item editor to simplify things. I'm getting reports from ArduPilot users for lack of support.

There is also the change to MAV_CMD_DO_SET_CAM_TRIGG_DIST.param3 which allows you to start triggering immediately instead of after the distance is reached. This allow you to create surveys which have better coverage at the incoming edge of the survey area. QGC always uses this for all the surveys it creates.

rmackay9, 2020-01-01:
@DonLakeFlyer, thanks for this.  I can't promise when we will implement this but we are trying to improve our gimbal and camera support.

peterbarker, 2024-07-31:
We now have support for those messages - and the handling of the take-picture-now MAV_CMD_DO_SET_CAM_TRIGG_DIST.param3

We got there... eventually :-)

closed 2024-07-31 by peterbarker

DonLakeFlyer, 2024-07-31:
Thanks for letting me know. Let me look if there are things in QGC I need to turn on now. @peterbarker Do you know what ArduPilot version these show up in. I probably need to do a version specific thing.