ArduPilot/ardupilot #13144: Logging failure in Copter 4.0 on 64Gb FAT32 SDCARD Durandal

andyp1per, 2020-01-01:
Logging failure in Copter 4.0 on 64Gb FAT32 SDCARD on Durandal
Card is not initialized

### Bug report

I have a 64GB SDCARD formatted using guiformat to FAT32. Copter 4.0 fails to initialize this on a Durandal. I have previously been able to use this card on a Kakute F7 AIO on master so it is either an issue with the Durandal or a new bug. No data appears on the card - no LASTLOG or TERRAIN folder so I guess the card cannot be written to. I also see this:

01/01/2020 13:40:08 : PreArm: Logging failed
01/01/2020 13:39:46 : PreArm: Logging failed
01/01/2020 13:39:38 : PreArm: Logging failed

A previously formatted and used 8GB card works fine.

**Version**
4.0

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Y6B

**Hardware type**

01/01/2020 13:38:57 : Frame: Y6
01/01/2020 13:38:57 : Durandal 0025001D 35375110 33383331
01/01/2020 13:38:57 : ChibiOS: 1ed2b2e9
01/01/2020 13:38:57 : ArduCopter V4.0.0 (49693540)

**Logs**

proficnc, 2020-01-01:
I’d suggest this is a feature request rather than a bug report.

Traditionally 4-16GB has been used with ardupilot.

andyp1per, 2020-01-01:
> 
> 
> I’d suggest this is a feature request rather than a bug report.
> 
> Traditionally 4-16GB has been used with ardupilot.

Nah :) I use high performance 64Gb cards on all my copters, this is the first time I have ever had a problem.

Safranek42, 2020-04-30:
Do you have any solution for this issue @andyp1per ? I'm also experiencing problem with logging on SD card with Durandal.