ArduPilot/ardupilot #13154: Kakute F7 Mini and Kakute F7 AIO PX4Flow Not Working in Copter-4.0

NitinJSanket, 2020-01-02:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

Optical flow gives no values in Kakute F7 Mini. Same thing works on Kakute F7 on Copter 3.6.12 though. Copter 4.0.0 has the same issue with kakute F7 too. Also, this is after the fix suggested in https://github.com/ArduPilot/ardupilot/issues/10105.

I also tried deleting all the code in AP_OpticalFlow.cpp to force it to use PX4Flow such that the only code remaining in void OpticalFlow::init(uint32_t log_bit) is backend = AP_OpticalFlow_PX4Flow::detect(*this); and backend->init();. No luck.

**Version**
4.0.0

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X4.

**Hardware type**
Kakute F7 Mini and Kakute F7 AIO.

**Logs**
Cant log on the Kakute F7 Mini. My Kakute F7 AIO SD card shield is broken.

Seeelefant, 2020-01-07:
Also Cheerson CX-OF Optical Flow is not working.

rmackay9, 2020-01-07:
We've found an issue with Copter-4.0.0 in which it isn't reporting the optical flow values to the ground station but the sensor itself works.  I'm woindering how you're testing that it doesn't work?  By the way, this should really be raised in the Copter-4.0 forums: https://discuss.ardupilot.org/c/arducopter/copter-40

closed 2021-08-15 by IamPete1