ArduPilot/ardupilot #13171: FrSky Telemetry on Kakute F7 not working as described

Seeelefant, 2020-01-04:
Bug report
FrSky Telemetry on Kakute F7 AIO not working with the setting mentioned here http://ardupilot.org/copter/docs/common-holybro-kakutef7aio.html . RCIN and Telemetry are both configured on UART6. Is there some kind of Inverter or dedicated cable missing?

Issue details

RX6 <-----> S.BUS
TX6 <-----> S.PORT

image
[image]

No telemetry data (by yapuu script)/no sensors dicovered.
Version
ArduCopter V4.0.0 (4969354)

Platform
[ ] All
[ ] AntennaTracker
[ X ] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

Airframe type
X4

Hardware type
Holybro Kakute F7 (not AIO)

Logs
02.01.2020 18:24:47 : PreArm: Battery 1 low voltage failsafe
02.01.2020 18:24:47 : PreArm: Battery 1 low voltage failsafe
02.01.2020 18:24:47 : PreArm: Compass not healthy
02.01.2020 18:24:17 : RCInput: decoding SBUS
02.01.2020 18:24:17 : PreArm: Battery 1 low voltage failsafe
02.01.2020 18:24:17 : PreArm: Battery 1 low voltage failsafe
02.01.2020 18:24:17 : PreArm: Compass not healthy
02.01.2020 18:24:17 : PreArm: Throttle below Failsafe
02.01.2020 18:23:53 : Frame: QUAD
02.01.2020 18:23:53 : KakuteF7 00340018 32365101 30383430
02.01.2020 18:23:53 : ChibiOS: 1ed2b2e9
02.01.2020 18:23:53 : ArduCopter V4.0.0 (4969354)

cglusky, 2020-02-02:
Have the same setup and tested to work with X8R and X4R.  Your params look fine.  This is a single wire/half duplex only so just SPort to TX6. 

Kakute will not power peripherals from USB and it will struggle to finish booting.  You need to have battery power applied.  Which means you should really remove props each time you want to plug into USB to make big changes.  I do anyway.  

I would also suggest you start a thread at discuss.ardupilot if it's not working as it's best place to get help.

Seeelefant, 2020-04-10:
Dear all, I tried it again with all recommendations listed, however I still do not how any success. However I use the recent firmware from FrSky https://www.frsky-rc.com/important-firmware-update-accst-d16/ , is there some kind of interference?

mattdholmes00, 2021-01-08:
seeelefant, any luck?  I too am having issues with a similar setup.  kakute f7 v1.5 (not the AIO version).  Interesting thing is that I'm successfully getting both the RX telemetry (not too exciting) AND the GPS telemetry, which indicates to me that the FC is actually passing something through properly.  But I'm not getting the other FC telemetry, e.g. battery/current etc.

details:

TX firmware on Taranis X9D+2019: ACCESS 2.3.10
RX firmware on R-XSR: ACCESS 2.1.0

Like you I have 
RX6 <-----> S.BUS
TX6 <-----> S.PORT

I also have: 
SERIAL6_BAUD=57
SERIAL6_OPTIONS=7
SERIAL6_PROTOCOL=10

My discovered sensors on the taranis include RSSI, RxBt, VFR, and GPS.  The first three come direct from the R-XSR RX, but the last one only started working when I configured the serial port for passthrough, so that makes me feel like I *almost* succeeded.

I have no errors in my logs.

I plug in the battery to test, props removed, to provide sufficient power.

I have not installed the Yaapu LUA yet, I was hoping to discover sensors before adding that layer.

mattdholmes00, 2021-01-08:
Update: I started installing the yaapu LUA script (haven't got it working fully) but then when I re-discovered telemetry sensors, they all showed up...

IamPete1, 2021-08-15:
sounds like this is resolved. I have a Kakute F7 somewhere where it was working.

closed 2021-08-15 by IamPete1