ArduPilot/ardupilot #13194: AP_Notify: Create automatic audible lost/crashed vehicle alarm

Pedals2Paddles, 2020-01-07:
### Feature request

**Is your feature request related to a problem? Please describe.**
When a vehicle crashes or is otherwise lost, it can be difficult to find it if you cannot see it.  If you do not have and RC connection, you cannot activate the lost vehicle alarm.  The lost vehicle alarm can only be activated by stick gesture or aux switch.  If you do not have a GCS connection and GPS, you cannot see its coordinates.

**Describe the solution you'd like**
- Lost vehicle alarm sound activate and stay activated when the RC and GCS are lost
- Lost vehicle alarm sound activate and stay activated on crash detection
- Change the lost vehicle alarm sound to something more unique, such as the sound of an ELT
- Change RGB lights to flash as bright white as possible and remain flashing after crash detection

**Describe alternatives you've considered**
Pop smoke and launch flares. But probably not practical at this time :)

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

krady21, 2020-02-24:
Is this still open?

Pedals2Paddles, 2020-02-24:
Yes

andyp1per, 2020-06-02:
Same as https://github.com/ArduPilot/ardupilot/issues/1023 for which there is an abandoned PR.

peterbarker, 2020-06-03:
@andyp1per I don't see an abandoned PR to cause the lost-vehicle alarm to sound on crash detection.  Pointers?

auturgy, 2020-06-03:
let's link to https://github.com/ArduPilot/ardupilot/issues/5856 and close one of them.

andyp1per, 2020-06-03:
@peterbarker this one: https://github.com/ArduPilot/ardupilot/pull/1997
Not quite the same, but close-ish

andyp1per, 2020-06-03:
@auturgy #5856 is asking for something different. This is simply about using the buzzer support we already have to signal a crashed vehicle's location.

andyp1per, 2020-06-03:
@Pedals2Paddles you self-assigned - does this mean you are working on it?

andyp1per, 2020-06-16:
I'm thinking that maybe the current lost vehicle stick command is sufficient for my needs since it's very difficult to tell whether you have crashed in acro.

peterbarker, 2020-06-17:
@andyp1per we turn crash detection off in Acro and Flip.

andyp1per, 2020-06-17:
@peterbarker - right. Even so, I suspect we could do a decent job of detecting that we are not moving in acro.

007harshmahajan, 2022-12-08:
I would like to work on this issue, is is open still ?

aimankhan24, 2023-03-27:
Is this issue still open? I would like to work on it.

rmackay9, 2023-03-27:
@aimankhan24,

Yes, the issue is still open.  best of luck.

peterbarker, 2026-11-15:
Duplicate of https://github.com/ArduPilot/ardupilot/issues/1023

closed 2026-11-15 by peterbarker