ArduPilot/ardupilot #13204: Fallback yaw support when GPS moving baseline yaw unavailable

tridge, 2020-01-08:
With support for easy to configure moving baseline yaw now in master we need to look at ways of handling outages of the GPS yaw value.
There are multiple parts to this:
 - a yaw source to prevent build-up of large Z gyro bias while waiting for initial RTK moving baseline lock. It can take 15 mins to get lock in some conditions and we've seen high levels of Z gyro drift in that time.
 - a fallback yaw source for when flying with GPS yaw as primary

For the first part I think we should fuse in a zero heading when disarmed with low movement (as indicated by accels) and setup for GPS yaw. So we'd have an assumption that the vehicle isn't rotating. That may be a poor assumption in some environments (eg. some ships), but we could also add a Z gyro bias reset when we get the initial GPS yaw to cope with ship environments.

For the fallback yaw source in flight I think we should learn bias values for the primary compass even when not using the compass. This would be similar to what we do with learning gyro biases for inactive gyros (which we added for better supporting IMU failure). The model it would use is the WMM, rotated by the GPS derived yaw and attitude estimate. We'd have a slow drift (time constant 30s or so?) of the compass bias values towards the value indicated by WMM and the current attitude. We'd also need to setup the mag variance states to a high level of uncertainty. The mag EF estimate would be set to the WMM field. The advantage of this scheme is that when we do lose GPS yaw the mag biases should already be consistent with the current attitude estimate of the vehicle, so we shouldn't get a big jump. We can then fallback to using the compass until we regain GPS yaw.
We'd need parameters to control this, perhaps an additional value for EK3_MAG_CAL that is "GPS yaw with compass fallback".

priseborough, 2020-01-09:
I've used the 'when on ground and not moving' method of constraining the yaw drift successfully in PX4/ecl - see https://github.com/PX4/ecl/blob/master/EKF/mag_fusion.cpp#L642-L657

priseborough, 2020-01-15:
A few thoughts

When on ground:

a) A movement detection test to determine isMoving. This would have a threshold similar to the movement test used by the startup gyro offset cal - could we use that test?
b) When isMoving = false, fuse the last INS yaw angle stored when isMoving = true 
c) When isMoving = true, fuse the current INS yaw angle (zero innovation) and store for use when movement stops (case b)
d) When GPS yaw becomes available, it might be safer to do a full filter reset. 

When in air:

a) We can go about 30 seconds without a yaw observation assuming gyro bias stability is better than 0.5 deg/sec
b) When using GPS yaw we could learn the mag bias errors to provide a backup yaw reference once  mag variances have stabilised and mag yaw has been consistent with EKF yaw for more than TBD seconds. Of course any fallback to mag should only be done when enabled by user parameter.

closed 2020-04-11 by tridge