ArduPilot/ardupilot #13215: AC_Avoidance: BendyRuler and Object Database should work in 3D

rmackay9, 2020-01-09:
BendyRuler's performance could (likely) be improved by enhancing both the [path planning algorithm ](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Avoidance/AP_OABendyRuler.h)and the [object database](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Avoidance/AP_OADatabase.h) to work in 3D.

The first step could be:

- extend the object database to store the altitude of each object instead of just it's XY position.  It may be useful to look at [this PR which actually removed the storage of the altitude](https://github.com/ArduPilot/ardupilot/pull/13024) because it wasn't being used
- when bendy ruler calculates the margin from objects it should calculate in 3D.  So the line should be a line in 3D space and the object's altitude should be part of the calculation

A follow-up step would be to modify BendyRuler so that when testing various paths, it doesn't simply test in many directions horizontally but also vertically.

This issue was raised after testing documented in this blog: https://discuss.ardupilot.org/t/bendyruler-with-lightware-sf40c-360-degree-lidar/50915

vishal-raj15, 2020-03-08:
@rmackay9  we can get the altitude value of an object from drone sensors , right ?

rmackay9, 2020-03-09:
@vishal-raj15,

yes, we should be able to use the orientation of the lidar and the distance to the obstacle along with the vehicle's attitude and position to calculate the altitude of the object along with the lat/lon (that we already calculate).

vishal-raj15, 2020-03-09:
@rmackay9  i would like to work on this issue

rmackay9, 2020-03-09:
@vishal-raj15,

Sure, you're very welcome to give it a try.  I'm not actively working on it myself but perhaps I can give some advice if you get stuck.

vishal-raj15, 2020-03-10:
@rmackay9  this issue require a 3d environment to show distance between objects and copter ?

rmackay9, 2020-03-11:
@vishal-raj15, that would be nice to have but I don't think it's required.  Doing that would require some enhancements to one of the simulators I think.. perhaps Morse or AirSim..

closed 2020-06-19 by rmackay9