ArduPilot/ardupilot #13229: Copter: release parachute if vehicle expectedly falls but maintains attitude

rmackay9, 2020-01-11:
This issue came up on the forums here: https://discuss.ardupilot.org/t/23kg-quad-drone-crash/49913 involving a situation where the vehicle had a mechanical failure and crashed but it maintained it's attitude during the fall which meant the parachute did not deploy.

We should enhance the parachute to deploy (perhaps optionally) if the vehicle loses it's ability to control altitude.

aercamti, 2020-02-27:
Here is another report of a multicopter dropping straight down: [https://discuss.cubepilot.org/t/x700-cube-2-1-mayday-mayday/2886]

Apparently no parachute was fitted, but simply the type of drop encountered here again underscores the necessity of this logic change being applied sooner than later.

This should ensure that no "tumble" is needed for deployment, but sudden, sustained drops should trigger chute and FTS.

Skycatpro, 2020-02-27:
We may have found this same (or other) situation where parachute was not ejected. We were doing specifically 4.0.2 parachute feature testing and last simulated motor failure caused crash. More about the case in here; https://discuss.ardupilot.org/t/copter-4-0-2-parachute-test-1-total-failure-2-success/52816/7

RickReeser, 2020-02-27:
Yes I was just going to add that discussion here. According to the onboard datalogger, it looks like the drone descended while mostly maintaining a level attitude.

Pedals2Paddles, 2020-02-28:
If the copter is well balanced and thrust loss is even, as in a lack of power rather than motor failure, straight down fairly level is a very reasonable scenario.  This happens frequently when a battery begins to go flat.  The motors are providing enough thrust to stabilize the attitude, but not enough to arrest a descent.

rmackay9, 2020-10-06:
This is resolved by PR https://github.com/ArduPilot/ardupilot/pull/15473 and will go out with Copter-4.1.  We should add some documentation of the CHUTE_SINK_RATE parameter to the parachute docs (see issue https://github.com/ArduPilot/ardupilot_wiki/issues/3066)

rmackay9, 2020-12-08:
This has been resolved in master so closing

closed 2020-12-08 by rmackay9