ArduPilot/ardupilot #13232: Loss of receiver behaviour change after 3.5.X to 3.6.X and 4.0.X

lthall, 2020-01-11:
On loss of connection between the transmitter and receiver when using SBUS the outputs would continue to be updated in 3.5.X.

In 3.6.X and newer the receiver outputs are not updated as soon as the failsafe flag was set.

I have come across this problem when a user updated from 3.5 that was using the receiver failsafe to define the behaviour during loss of radio connection and had disabled radio failsafe on the autopilot. The result is the aircraft continues as if the last rc input was valid and can only be controlled using a telemetry link.

I do not know what the correct response is to this change in behaviour as I understand the arguments for keeping the last valid radio inputs before the receiver failsafe bit was set.

lthall, 2020-01-11:
After considering the options I believe that keeping the current values is the correct approach. As such I would be happy to simply close this issue.

I would suggest that we should clearly document the change in behaviour however.

There is an option to not set the throttle failsafe. In this instance it may be prudent to also set the "Ignore Receiver Failsafe Bit" flag.

IamPete1, 2021-09-12:
@lthall Should we close this? Is there anything else we need to to do?

closed 2022-07-08 by lthall