ArduPilot/ardupilot #13233: Help needed in Developing New Controller Options other than PID

aby03, 2020-01-11:
#### Issue details
_Please describe the problem, or desired feature_
I need to try out some other torque based controllers for my copter other than PID controller. Could you give me some basic outline how can i go about it? I need to implement it for copter specifically. 

#### Platform
[  ] Copter

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
X quad

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
pixhawk

lthall, 2020-01-11:
This is not where you ask for support. Please take this to the discuss forums.

closed 2020-01-11 by lthall

rmackay9, 2020-01-13:
Hi @aby03 

the user support forums are here: https://discuss.ardupilot.org/c/arducopter/copter-40
developers (which it seems like you are) are also welcome to chat here: https://gitter.im/ArduPilot/ardupilot