ArduPilot/ardupilot #13236: Add External Flight Termination Feature to Autdupilot per ASTM F3322-18

eengelgau, 2020-01-11:
Hi Everyone, for background on this feature request see:

https://discuss.ardupilot.org/t/add-external-flight-termination-feature-to-autdupilot-per-astm-f3322-18/50976

Right now I'm not completely syre all this would affect.  The Copter code base for the autopilot.  Also Planes need this feature as well.

**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

-Gene

khancyr, 2020-01-11:
Hello,

We are already compliant for long time.
You just need to set an output port as input and set the function either to flight_termination/emotor_stop or parachute_release.

eengelgau, 2020-01-11:
Hi, That would be awesome.  So I'm looking for this and I can't find this in the documentation?  So while it's possible where are these settings?  Can you or someone screen shot the setup screen perhaps using Aux 5 as the Flight Termination input, PWM, 2000 us pulse to trigger?

Thanks for the info so far,

Gene

Pedals2Paddles, 2020-01-12:
ArduPilot can detect the condition (falling out of the sky) or take a manual RC input which will stop the motors and trigger an output to pop the chute.

What ArduPilot currently does not have is way for an external input other than the RC receiver to trigger the emergency stop.  So if you have an external parachute device, that does the detection and chute popping, there is no way to tell ArduPilot to stop the motors right now.

Pedals2Paddles, 2020-01-12:
Taking a stab at this here: https://github.com/ArduPilot/ardupilot/pull/13238

eengelgau, 2020-02-26:
Hi Everyone, I had opened this issue about 6 weeks ago, and it seem like for a time that a solution may come.  Has the ability to do Flight Termination (stop rotors) via an external Aux PWM input been implemented?  If to what is the status, or branch it is in.  What needs to be downloaded.

This right now is a limitation for conforming to ASTM F3322-18 certification until this is completed.

See more information here: https://discuss.ardupilot.org/t/add-external-flight-termination-feature-to-autdupilot-per-astm-f3322-18/50976

Thanks!!

-Gene

RickReeser, 2020-02-27:
As I understand, the ASTM standard requires that the FTS be completely independent from the flight controller. I suppose the reasoning is that the system must be able to handle the situation where the FC locks up but still outputs PWMs. This would still be a useful feature for Ardupilot, but it wouldn't strictly conform to the standard unless we can somehow guarantee that will never happen.

eengelgau, 2020-02-27:
Hi Rick, your right, this is an imperfect solution.  Overriding the ESC's
is ideal with a emergency stop signal.  But getting all the
manufacturers of the ESC, which are all in China (as far as I know), is
going to be tough.  DJI on it's own has done nothing to make Flight
Termination any easier on it's drones.  Getting this to happen on all the
various ESC manufacturers is even harder.

So in this case, and to just get something working with the least
resistance, this seems like a good step.  Make since?

As I said PX4 code base has this now as of the start of the year.  So
hopefully someone in the Ardupilot community can take this up soon.

-Gene

- Regards
Gene Engelgau
Web: http://fruitychutes.com - Professional Aerospace Recovery Systems
Facebook: facebook.com/fruitychutes
Twitter: @fruitychutes
Cell: 408-499-9050