ArduPilot/ardupilot #13239: Parameter RNG_FND1_PIN does not get migrated during update to AC4.0

mtbsteve, 2020-01-12:
### Bug report

**Issue details**
Parameter does not get migrated during update to AC4.0

_Please describe the problem_
During the update from AC3.6.10 to AC4.0.0, the paramter RNG_FND1_PIN, which I had set to 54 was set to -1 after update.

**Version**
_What version was the issue encountered with_ AC4.0.0

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quad
**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Cube 2.1
**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2020-01-13:
I tried to reproduce this but wasn't able to sadly.  I did this:

- loaded Copter-3.6.12 onto the vehicle and reset all parameter to their defaults using the "Reset to Default" button the MP's Full Parameter Tree page
- Re-connected and set RNGFND_TYPE = 1, RNGFND_PIN = 66.
- Upgraded the firmware to Copter-4.0.1-rc2 and found the parameter was converted correctly (i.e. RNGFND1_PIN = 66)

There definitely were some issues with the parameter conversion found during beta testing but I think I [fixed all these issues](https://github.com/ArduPilot/ardupilot/pull/12922) before the Copter-4.0.0 release went out.

mtbsteve, 2020-01-13:
I just retested the upgrade from AC 3.6.12 to AC 4.0.1 RC2 and could not reproduce this issue. I am closing this for now.

closed 2020-01-13 by mtbsteve

rmackay9, 2020-01-14:
Great, thanks for the followup!