ArduPilot/ardupilot #13240: Problems with FRSky Pass Through test

andyp1per, 2020-01-12:
**Issue details**

I am having significant problems with the FRSky PassThrough test. It always overflows the stack for me and never passes. I am in my FFT branch https://github.com/ArduPilot/ardupilot/pull/11886
but get the same effect if I complete turn off the FFT support at compile time. This makes me think there is something more fundamentally wrong going on here

**Version**
master
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
PLane

**Hardware type**
SITL

**Logs**
 The error is basically:
```
FRSkyPassthrough: Sending poll
AUTOTEST: Checking 20485
AUTOTEST: validating vel_and_yaw(0xdd40000)
AUTOTEST:  yaw=3540 gpi=35394
AUTOTEST:  xy_vel=0
AUTOTEST:  z_vel_dm_per_second=0
Still wanting (0x5005)
FRSkyPassthrough: Battery 1 status (0x5003)=126
AUTOTEST: Checking 20485
AUTOTEST: validating vel_and_yaw(0xdd40000)
AUTOTEST:  yaw=3540 gpi=35394
AUTOTEST:  xy_vel=0
AUTOTEST:  z_vel_dm_per_second=0
Still wanting (0x5005)
```

peterbarker, 2020-01-12:
@andyp1per That output you've pasted isn't a sign of anything failing.  It's just rather verbose.

If the output *stops* there, then that is a problem :-)

Copying this in from the other PR:

>  *  e2728ba AP_Frsky_Telem: thread overflows stack in SITL

If this has any effect then we have a subtle bug somewhere in here:

https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_HAL_SITL/Scheduler.cpp#L294

We enforce a minimum thread stack size of "a rather large number" in SITL.

Further - why would this stack overflow simply exist on SITL and not on any other board?  Can you explain where the stack overflow occurs?

Note that there *is* something weird going on with Frsky telem.  tridge and I believe there's something in the way we're handling `pthread_attr_setstack`
which Isn't Quite Right - semantics on that call regarding guard pages have changed over time, apparently.

andyp1per, 2020-01-12:
Increasing the stack definitely stops it falling over on macOS and windows. The test still fails however. I had similar stack problems with the fft in SITL - seems weird.

andyp1per, 2020-01-12:
Sorry - yes it stops there and the test fails

peterbarker, 2020-01-12:
@andyp1per If you're starting a thread then it's probably related.

@tridge this Frsky stuff looks a little bodgy.  I can't see a bug here, but the fact we're calling all sorts of methods before `init()` is called is rather unfortunate.  See https://github.com/peterbarker/ardupilot/commit/dfd546c9d39df433f5ad2d56e8d94e795553cb7f

andyp1per, 2020-01-13:
> If this has any effect then we have a subtle bug somewhere in here:

Actually @peterbarker this is easy to understand - we don't *actually* overflow the stack, we simply overflow our *virtual* stack pointer which we determine through a magic number. The virtual pointer is constrained by the stack size passed into thread creation.

andyp1per, 2020-01-13:
> Further - why would this stack overflow simply exist on SITL and not on any other board? Can you explain where the stack overflow occurs?

64bit vs 32bit maybe? Seems like there are a ton of differences between a modern multicore intel chip running macos and a STM32 running ChibiOS - seems inevitable that the stack usage is going to be bigger.

andyp1per, 2020-01-13:
@peterbarker isn't your logic wrong:

> AUTOTEST: validating vel_and_yaw(0xdd40000)
> AUTOTEST:  yaw=3540 gpi=35394

clearly yaw and gpi are equivalent here, but your logic:
if int(round(yaw/10,3)) == int(round(gpi.hdg/100, 3)):

fails

andyp1per, 2020-01-13:
@peterbarker because you did such a great review of my runcam stuff I have fixed your maths for you ;) https://github.com/ArduPilot/ardupilot/pull/13253/commits/718d872d26e42017a2da617f5836d3a6ba47e7a3

andyp1per, 2020-01-13:
There's still a problem with FRSkyD - I can't repro this locally so not sure what I can do. See https://github.com/ArduPilot/ardupilot/pull/13253

peterbarker, 2020-01-14:
I'm not convinced it's a conventional stack overflow.  You attempts to massively increase the stack size showed that, I think.

Try commenting out the call to pthread_attr_setstack and see if that makes the problem go away for you, please?

andyp1per, 2020-01-14:
> I'm not convinced it's a conventional stack overflow. You attempts to massively increase the stack size showed that, I think.
> 
> Try commenting out the call to pthread_attr_setstack and see if that makes the problem go away for you, please?

Hey @peterbarker - sorry perhaps I wasn't clear - It definitely *not* a conventional stack overflow
https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_HAL_SITL/Scheduler.cpp#L351 checks stack_min which is the virtual stack size passed into https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_HAL_SITL/Scheduler.cpp#L285 and totally different to https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_HAL_SITL/Scheduler.cpp#L294 which is the *actual* stack size and much bigger. So setting pthread_attr_setstack will have zero effect as that changes alloc_stack, not stack_min.

andyp1per, 2020-01-14:
But @peterbarker & @tridge regardless - I don't think this is now a stack problem with my change. The FRSkyD fails and it appears to fail because it makes no progress *and* simulation time makes no progress (because the test timeout never triggers) - that indicates to me that something has locked the main thread, which seems pretty bad.

peterbarker, 2020-01-14:
On Mon, 13 Jan 2020, Andy Piper wrote:

> But @peterbarker & @tridge regardless - I don't think this is now a stack problem with my change. The FRSkyD fails and it appears to fail because it makes
> no progress and simulation time makes no [...]

It's almost certainly crashed, complaining that the malloc metadata is 
corrupt.

andyp1per, 2020-01-14:
@peterbarker does that mean you think it is still stack related?

peterbarker, 2020-01-14:
On Tue, 14 Jan 2020, Andy Piper wrote:

> @peterbarker does that mean you think it is still stack related?

Yes.  Did you try simply commenting out the pthread_attr_setstack call?

andyp1per, 2020-01-14:
@peterbarker I can't reproduce it locally but have pushed a change to the PR to see if travis can

andyp1per, 2020-01-14:
@peterbarker I have tried the test with ASAN and there are no reported problems. Generally ASAN is pretty good at picking up malloc corruption.

andyp1per, 2020-01-14:
Does valgrind work in Vagrant? I get this:
```
Running: "valgrind" "--soname-synonyms=somalloc=nouserintercepts" "--vgdb-prefix=/tmp/vgdb-pipe" "-q" "--log-file=arduplane-plane-elevrev-valgrind.log" "/vagrant/build/sitl/bin/arduplane" "-w" "-S" "--home" "-35.362938,149.165085,585,354" "--model" "plane-elevrev" "--speedup" "8" "--defaults" "/vagrant/Tools/autotest/default_params/plane-jsbsim.parm"
==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_e9e0f184
==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_6e753d6d
==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_eff08da2
==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_21dfda33
==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_573da1f0
==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_1c354369
==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_096de1ee
==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_aab3e68f
==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_2072ed1c
==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_54d5ad25
==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_895a2efa
valgrind: Startup or configuration error:
valgrind:    Can't create client cmdline file in /vagrant/tmp/valgrind_proc_10142_cmdline_895a2efa
valgrind: Unable to start up properly.  Giving up.
>>>> FAILED STEP: fly.ArduPlane.FRSkyD at Tue Jan 14 23:21:45 2020 (End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7fcb01153990>
command: /usr/bin/valgrind
args: ['/usr/bin/valgrind', '--soname-synonyms=somalloc=nouserintercepts', '--vgdb-prefix=/tmp/vgdb-pipe', '-q', '--log-file=arduplane-plane-elevrev-valgrind.log', '/vagrant/build/sitl/bin/arduplane', '-w', '-S', '--home', '-35.362938,149.165085,585,354', '--model', 'plane-elevrev', '--speedup', '8', '--defaults', '/vagrant/Tools/autotest/default_params/plane-jsbsim.parm']
buffer (last 100 chars): ''
before (last 100 chars): 'grant/tmp/valgrind_proc_10142_cmdline_895a2efa\r\nvalgrind: Unable to start up properly.  Giving up.\r\n'
after: <class 'pexpect.exceptions.EOF'>
```

andyp1per, 2020-01-14:
Ok was missing a tmp directory. With that fixed the only valgrind output I get is this which looks unrelated:
```
==10179== Conditional jump or move depends on uninitialised value(s)
==10179==    at 0x40A8B1: Plane::convert_mixers() (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x40A747: Plane::load_parameters() (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x41C731: Plane::init_ardupilot() (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x4050CC: Plane::setup() (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x4C4F2F: HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x405D0B: main (in /vagrant/build/sitl/bin/arduplane)
==10179== 
==10179== Conditional jump or move depends on uninitialised value(s)
==10179==    at 0x40A8FB: Plane::convert_mixers() (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x40A747: Plane::load_parameters() (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x41C731: Plane::init_ardupilot() (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x4050CC: Plane::setup() (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x4C4F2F: HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x405D0B: main (in /vagrant/build/sitl/bin/arduplane)
==10179== 
==10179== Conditional jump or move depends on uninitialised value(s)
==10179==    at 0x40A796: Plane::load_parameters() (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x41C731: Plane::init_ardupilot() (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x4050CC: Plane::setup() (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x4C4F2F: HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const (in /vagrant/build/sitl/bin/arduplane)
==10179==    by 0x405D0B: main (in /vagrant/build/sitl/bin/arduplane)
==10179==
```

andyp1per, 2020-01-14:
Ok amazingly the plane tests pass on travis with the setstacksize commented out:
https://travis-ci.org/ArduPilot/ardupilot/jobs/637156243?utm_medium=notification&utm_source=github_status
Your move @peterbarker 😄

peterbarker, 2020-01-15:
On Tue, 14 Jan 2020, Andy Piper wrote:

> ==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_e9e0f184

Could you strace that to get an errno, please?

peterbarker, 2020-01-15:
@andyp1per which Vagrant VM was that on?

Which branch?

What was the state of the eeprom.bin - fresh and clean or had it been used before?

andyp1per, 2020-01-15:
> On Tue, 14 Jan 2020, Andy Piper wrote: ==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_e9e0f184
> Could you strace that to get an errno, please?

The error is gone - I did ```mkdir /vagrant/tmp```

andyp1per, 2020-01-15:
> @andyp1per which Vagrant VM was that on?
> 
> Which branch?
> 
> What was the state of the eeprom.bin - fresh and clean or had it been used before?

bionic64
I just ran autotest with the FRSkyD test - does that clean eeprom.bin?

peterbarker, 2020-01-15:
On Wed, 15 Jan 2020, Andy Piper wrote:

>       On Tue, 14 Jan 2020, Andy Piper wrote: ==10142== VG_(mkstemp): failed to create temp file: /vagrant/tmp/valgrind_proc_10142_cmdline_e9e0f184
>       Could you strace that to get an errno, please?
> 
> The error is gone - I did mkdir /vagrant/tmp

Thanks.  Sounds like something we need to fix; either create it as part of 
bringup or pointing somewhere else.

Actually, I'm a little surprised /vagrant/tmp actually worked; I thought 
we put Valgrind temporary files in /tmp as Valgrind was creating named 
pipes on the /vagrant "bind" mount and that didn't work out too well....

peterbarker, 2020-07-10:
@andyp1per I think these tests are running cleanly now, yes?

Did you get Valgrind running properly?

andyp1per, 2020-07-11:
@peterbarker can't say I've seen this fail in CI for a long time and I haven't tried locally for a long time

peterbarker, 2020-07-11:
@andyp1per Fixed until someone says otherwise then - thanks!

closed 2020-07-11 by peterbarker