ArduPilot/ardupilot #13242: Copter: add check if only one of MOT_PWM_MIN or MOT_PWM_MAX is set

rmackay9, 2020-01-13:
This issue is raised in response to this report on the forums which involved a user setting the MOT_PWM_MIN parameter (presumably to ensure the output to the motors was never below this value) but failed to set the MOT_PWM_MAX value.

We should add a pre-arm check to ensure that both these valures are either zero or non-zero.  I.e. a situation where only one of them is non-zero should trigger a pre-arm failure.

kalaiselvan1219, 2020-01-13:
It's on only AC 4.0.0 stable version or will apply for all AC versions.?

kd0aij, 2020-01-13:
link to report?

rmackay9, 2020-01-13:
Ah, right, here's the link to the discussion. https://discuss.ardupilot.org/t/copter-4-0-did-great-thou-scary-with-hexa-failed-motor/50808/9

In the end the user's issues are not related to this misconfiguration but we should still add the check

PR #13363 (2020-01-29): Copter: check if only one of MOT_PWM_MIN or MOT_PWM_MAX is set
fixes #13242 

Pre-arm check to ensure that both MOT_PWM_MIN and MOT_PWM_MAX values are either zero or non-zero. When only one of them is non-zerol trigger a pre-arm failure.
[image]
[image]
closed 2020-01-29 by rmackay9