ArduPilot/ardupilot #13251: wp_nav requested speed reduce can cause attitude shock

lukezhqin, 2020-01-13:
### Bug report

**You guys would easily met this issue if you change one of your waypoint speed and reduce the request speed in the mission. Then after this do_cmd waypoint take effect in SITL or real filght, the vehicle would definitely make a sudden brake just like the following figure shows.**
[image]
[image]

**I konw it's better to try to update the '_leash' in pos_control.cpp but I actually haven't find a proper way yet. Now, my method in this PR may not look elegant enough but it can indeed avoid this waypoint change to reduce speed issue from distance limitation check of '_leash' once we update the speed request from the mission planned.**

**Version**
_tag Copter-4.0.0, master branch and all versions and branches before_

**Platform**
[  ] All
[  ] AntennaTracker
[×] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_hex, Y6, octa etc_

**Hardware type**
_SITL, Pixhawk, Cube, Pixracer, Navio2, etc_

**Logs**
_Please provide a link to any relevant logs that show the issue_

lukezhqin, 2020-01-13:
@tridge @rmackay9 Please help review this issue and PR.

bnsgeyer, 2020-02-10:
I believe this is the same as issue #11048 and proposed fix is PR #12821

rmackay9, 2020-02-21:
One of the two possible fixes has been merged to master and should be released with Copter-4.0.4 in a month or so.  Thanks very much for raising this issue!

closed 2020-02-21 by rmackay9