ArduPilot/ardupilot #13252: RNGFND1_OFFSET value is ignored

mtbsteve, 2020-01-13:
**Bug report**

**Issue details**
With AC 4.0 the parameter RNGFND1_OFFSET is being ignored. I am using a LidarLite connected via PWM. It works well with Arducopter 3.6 all versions with the NuttX core.

_Please describe the problem_
Here are 2 screenshots of the rangefinder output, the first one with AC 3.6 with the correct altitude and RNGFND_OFFSET=-20 
[image]

and the second screenshot with AC 4.0.1 RC2 and RNGFND1_OFFSET set to -20.
Whatever I enter is ignored. Seems like its being bissed after renaming of the parameters in AC 4.
[image]

Another observation is the voltage, with AC 3.6 the voltage is displayed, with AC4 there is no reading.

**Version**
_What version was the issue encountered with_
AC 4.0.1 RC2, AC 4.0.0

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quad
**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Cube green
**Logs**
_Please provide a link to any relevant logs that show the issue_
[https://1drv.ms/u/s!AnKeW8KMoCcyhiMb9kF6WwfbQgGA?e=sqpNtV](url)

rmackay9, 2020-01-16:
I've confirmed the issue and it seems to be that this is a regression from Copter-3.6.x.

In fact, this parameter is not used by the majority of drivers.  In Copter-4.0 only the analog driver uses it.  In Copter-3.6 it was the analog driver and the PWM driver that used it.

I think we should fix this in two steps:

1. restore the PWM driver's use of this parameter
2. add the use of this parameter to all other drivers

This report was also raised in [this discussion](https://discuss.ardupilot.org/t/rngfnd1-offset-not-working-for-pwm-and-i2c-lidarlite/51125)

rmackay9, 2020-02-01:
This is fixed for the PWM range finders and is [included in Copter-4.0.2-rc1](https://discuss.ardupilot.org/t/copter-4-0-2-rc1-available-for-beta-testing/51790) so I'll close it here.  Thanks!

closed 2020-02-01 by rmackay9