ArduPilot/ardupilot #13257: ROVER?: Compass Calibration silently fails if enabled compass is not present

jmachuca77, 2020-01-13:
### Bug report

**Issue details**
When using only a UAVCAN compass (USE_COMPASS 0, USE_COMPASS2 0, USE_COMPASS3 1, COMPASS_PRIMARY 3), and the compass is not detected, Mag Calibration does not start, but there is no indication anything failed.

**Version**
ROVER 4.0.0 release 

**Platform**
Not sure if this is only on Rover, might be on other platforms too.
[  ] All
[  ] AntennaTracker
[  ? ] Copter
[ ? ] Plane
[ X ] Rover
[ ? ] Submarine

**Hardware type**
Tested using Here2 on UAVCAN with CubeBlack and CubeOrange

jmachuca77, 2020-01-13:
Upon further testing this happens also if the only compass enabled is an external compass. Even if it is on I2C and it is correctly detected.

jmachuca77, 2020-01-13:
This TLOG shows an attempt to start compass calibration with only the external compass enabled.

[2020-01-13 12-24-43.tlog.zip](https://github.com/ArduPilot/ardupilot/files/4055388/2020-01-13.12-24-43.tlog.zip)

WickedShell, 2020-01-13:
We just need to send a NACK here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass_Calibration.cpp#L116 if no calibrations were started, as well as NACK here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass_Calibration.cpp#L99

jmachuca77, 2020-01-13:
This TLOG shows how after enabling an internal compass and starting calibration it starts correctly.

[2020-01-13 12-29-13.tlog.zip](https://github.com/ArduPilot/ardupilot/files/4055433/2020-01-13.12-29-13.tlog.zip)

tridge, 2020-03-31:
note that in these logs COMPASS_ENABLE=0

rmackay9, 2020-03-31:
*maybe* we don't display a message to the user when they try to do a compass calibration when compass-enable = 0.

closed 2026-03-16 by peterbarker