ArduPilot/ardupilot #13258: EKF variance and drone crash using SITL + Gazebo

phamelin, 2020-01-13:
#### Issue details
I followed the instructions to use the Gazebo plugin with ArduCopter SITL [1]. First, the EKF behavior is not stable, e.g. long delay before GPS fix, attitude inconsistency, etc. When all PreArm checks are finally ok, I try to takeoff using mavproxy:

```
mode GUIDED
arm throttle
takeoff 10
```
The copter actually takeoff but quickly flip over and crash on ground. It seems that I'm not the only one having this issue [2].

[1] https://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html
[2] https://discuss.ardupilot.org/t/sitl-with-gazebo-drone-crashes/26743

#### Version
ArduCopter master (commit bc7e15a)
Gazebo 9
ArduCopter Gazebo plugin: https://github.com/SwiftGust/ardupilot_gazebo

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
quad

#### Hardware type
SITL + Gazebo 9

#### Logs

```
APM: Preparing log system
APM: Prepared log system
online system 1
Mode STABILIZE
AP_IRLock_SITL::init()
APM: ArduCopter V4.0.1-dev (bc7e15ab)
APM: 14e2eb399adf4bbda591e4f099a63d26
APM: Frame: QUAD
APM: Barometer 1 calibration complete
Init Gyro***

Ready to FLY APM: GPS 1: detected as u-blox at 115200 baud
APM: EKF2 IMU0 initial yaw alignment complete
APM: EKF2 IMU1 initial yaw alignment complete
APM: EKF2 IMU1 tilt alignment complete
APM: EKF2 IMU0 origin set
APM: EKF2 IMU1 origin set
APM: PreArm: EKF attitude is bad
APM: EKF2 IMU0 tilt alignment complete
Flight battery 100 percent
APM: EKF2 IMU1 is using GPS
Got MAVLink msg: COMMAND_ACK {command : 11, result : 0}
Mode GUIDED
APM: PreArm: Need 3D Fix
Got MAVLink msg: COMMAND_ACK {command : 400, result : 4}
APM: PreArm: Need 3D Fix
APM: EKF2 IMU0 is using GPS
APM: Arming motors
Got MAVLink msg: COMMAND_ACK {command : 400, result : 0}
ARMED
Got MAVLink msg: COMMAND_ACK {command : 22, result : 0}
APM: EKF2 IMU1 in-flight yaw alignment complete
APM: EKF primary changed:1
APM: EKF2 IMU0 ground mag anomaly, yaw re-aligned
APM: EKF primary changed:0
APM: GPS Glitch
APM: Potential Thrust Loss (2)
APM: EKF variance
Mode LAND
```

phamelin, 2020-01-13:
I would like to add that this problem is only with the SwiftGust gazebo plugin (https://github.com/SwiftGust/ardupilot_gazebo). I'm not sure what is the difference with the Khancyr plugin (https://github.com/khancyr/ardupilot_gazebo), which actually seems to work better?

varunvp, 2020-11-27:
I'm facing the same issue with both khancyr and swiftgust plugins. I'm running ArduCopter 3.6.10, and this happens when I launch the world with gazebo_ros. Just running it from gazebo doesn't give any problems. Looks like I'll have to switch to PX4 for simulations.