ArduPilot/ardupilot #13294: PreArm check text overflow mavlink buffer

tecnic08, 2020-01-16:
### Bug report

**Issue details**
Low battery prearm text is longer than the size of the mavlink text buffer. Causing a missing null terminator.

I'll create a PR soon.

**Version**
3.6.12

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad-X

**Hardware type**
Pixhawk 4 (fmuv5), Pixhawk 2 (fmuv3)

**Logs**
No log, but photo should be suffice.

[image]

rmackay9, 2020-01-16:
I thought that [STATUSTEXT](https://mavlink.io/en/messages/common.html#STATUSTEXT) messages didn't need to be null terminated if they were exactly 50 characters long

The "description" says, "Status text message, without null termination character".

tecnic08, 2020-01-16:
> I thought that [STATUSTEXT](https://mavlink.io/en/messages/common.html#STATUSTEXT) messages didn't need to be null terminated if they were exactly 50 characters long
> 
> The "description" says, "Status text message, without null termination character".

I'm sorry, ignore the part that I say null terminator. However, the  "PreArm: PreArm:..." is causing the string to be longer than 50 characters.

tecnic08, 2020-01-16:
PR submitted: https://github.com/ArduPilot/ardupilot/pull/13295

rmackay9, 2020-02-25:
Resolved in Copter-3.6 branch (although it's unlikely to be released) and master so closing.

closed 2020-02-25 by rmackay9