ArduPilot/ardupilot #13298: Cannot build ArduCopter with waf

azhdarshirinzada, 2020-01-16:
### Bug report

**Issue details**

cannot build arducopter with command 
./waf configure --board CubeBlack

until this every step completed fine

**Logs**
./waf configure --board CubeBlack
Setting top to                           : C:\cygwin64\home\[User]\ardupilot
Setting out to                           : C:\cygwin64\home\[User]\ardupilot\build
Autoconfiguration                        : enabled
Setting board to                         : CubeBlack
Using toolchain                          : arm-none-eabi
Could not find the program ['arm-none-eabi-ar']
(complete log in C:\cygwin64\home\[User]\ardupilot\build\config.log)

this is config.log file:

# project  configured on Thu Jan 16 12:52:55 2020 by
# waf 2.0.9 (abi 20, python 30800f0 on win32)
# using C:\cygwin64\home\Azhdar Shirinzade\ardupilot\modules\waf\waf-light configure --board CubeBlack
#
----------------------------------------
Setting top to
C:\cygwin64\home\Azhdar Shirinzade\ardupilot
----------------------------------------
Setting out to
C:\cygwin64\home\Azhdar Shirinzade\ardupilot\build
----------------------------------------
Autoconfiguration
enabled
----------------------------------------
Setting board to
CubeBlack
----------------------------------------
Using toolchain
arm-none-eabi
from C:\cygwin64\home\Azhdar Shirinzade\ardupilot: Could not find the program ['arm-none-eabi-ar']

rmackay9, 2020-01-16:
@azhdarshirinzada,

Of course you've installed the gcc compiler as mentioned on the wiki here? https://ardupilot.org/dev/docs/building-setup-windows-cygwin.html#install-the-gcc-compiler

I know we recently changed the compiler so I wonder if we also need to update the compiler here: https://firmware.ardupilot.org/Tools/STM32-tools/

I'm afraid I'll need to ask @tridge or @peterbarker

azhdarshirinzada, 2020-01-17:
Yes, already installed.

[image]

Sobhit25, 2021-01-19:
I am facing the same issue on ubuntu

khancyr, 2021-01-19:
@Sobhit25 hello, please ask support on https://discuss.ardupilot.org/ . 
The Ubuntu installation is already well documented and tested https://ardupilot.org/dev/docs/building-setup-linux.html so double check that you have followed every step.
I am closing this as we have push numerous fix in our installation scripts and this belong to discuss anyway.

closed 2021-01-19 by khancyr

mikeh9, 2022-01-17:
The problem is that you need to logout / login after you run the Tools/environment_install/install-prereqs-ubuntu.sh -y

or you can type:
. ~/.profile