ArduPilot/ardupilot #13324: MAV_CMD_SET_MESSAGE_INTERVAL (511 ) don't work with interval of 40 ms or less

Eli-Pinchover, 2020-01-18:
### Bug report

Request position message using MAV_CMD_SET_MESSAGE_INTERVAL (511 ) doesn't respond within the required interval.

Sending message COMMAND_LONG ( #76 ) with MAV_CMD MAV_CMD_SET_MESSAGE_INTERVAL (511 ) to request a position message (#33) at intervals of 40 ms ( down to 20 ms) doesn't respond coherently.

**Version**
Copter 3.6.12 
Copter 4.0.0

**Platform**
[*  ] Copter

**Hardware type**
Pixhawk

peterbarker, 2020-01-19:
On Sat, 18 Jan 2020, Eli-Pinchover wrote:

> Request position message using MAV_CMD_SET_MESSAGE_INTERVAL (511 ) doesn't respond within the required interval.
>
> Sending message COMMAND_LONG ( #76 ) with MAV_CMD MAV_CMD_SET_MESSAGE_INTERVAL (511 [...]

We test this in autotest - so it should be working.

What's your main loop rate?  We limit statustext output to 80% of the main 
loop rate - so if you are running at 50Hz you can't get 20ms intervals.

Past that - check your bandwidth limits; if you're using standard SiK 
radios you may be limited there.

Peter

Eli-Pinchover, 2020-01-21:
> On Sat, 18 Jan 2020, Eli-Pinchover wrote: Request position message using MAV_CMD_SET_MESSAGE_INTERVAL (511 ) doesn't respond within the required interval. Sending message COMMAND_LONG ( #76 ) with [...]

the main loop is 400 Hz.
I run the ardupilot code with no changes on pixhawk (mR pixhawk).
the pixhawk is connected to raspberry pi 3 using 1Mb serial channel (telem2 configured as 921)
and mavproxy send the messages to another pc  on the same LAN.
However, the requested messages dont arrive at reqested interval ( 20, 30 40 ms)
I sent alse  MAV_CMD MAV_CMD_REQUEST_MESSAGE (512 )  every 20 (30, 40) ms but didn't get response for every single request.
What I need is getting the  position message at a rate of 50 hz. I don't get it with the downloaded code. Should I write my own task?

rmackay9, 2020-01-22:
I wonder if it's possible that some other process (like mavproxy) is sending request-data-stream messages that are overriding the message-interval requests.

peterbarker, 2020-01-22:
[image]

Works For Me™

I don't think you can expect this to work on 3.6.12 - pretty sure it was new for 4.0

peterbarker, 2020-01-22:
(should have mentioned; that's a PixRacer connected via USB)

Eli-Pinchover, 2020-01-22:
> (should have mentioned; that's a PixRacer connected via USB)

Thanks a lot for your answers.
I'll check again, making sure to run ardupilot ver 4.x
I use serial channel telem2  at 1Mb. I think this is good enough for my requirements.
mR pixhawk vs. PixRacer -  can this be the reason?
Eli.

Eli-Pinchover, 2020-01-22:
> I wonder if it's possible that some other process (like mavproxy) is sending request-data-stream messages that are overriding the message-interval requests.
The mavproxy is just a proxy. It is just  forwarding the messages coming  from the pixhawk.

Eli-Pinchover, 2020-01-22:
> (should have mentioned; that's a PixRacer connected via USB)

Thanks a lot for your nice answers.
I'll check again using ardupilot ver 4.x.
I use the telem2 1Mb channel - seem to me that this bandwidth  is good enough.
mR pixhawk vs. PixRacer  - could this be the reason?

Eli.

peterbarker, 2020-01-23:
No difference between Pixhawk and PixRacer in this case.

Eli-Pinchover, 2020-01-28:
> [image]
> 
> Works For Me™
> 
> I don't think you can expect this to work on 3.6.12 - pretty sure it was new for 4.0

I've checked again the issue on raspberry-pi  <====>  pixhawk (1Mb channel - telem2)
I commanded same command as you, "long SET_MESSAGE_INTERVAL 33 10000" ,  
 "long SET_MESSAGE_INTERVAL 33 5000"
I got graphs seen as yours. But, chacking carefully Wireshark pcap captures, shows that the messages don't arrive every 10  or 5  ms.  Some are 3 ms diff,  and some are  30 ms diff.
Attach pcap files for both commands showing this.
[rpi-mp-long.zip](https://github.com/ArduPilot/ardupilot/files/4124614/rpi-mp-long.zip)

peterbarker, 2020-01-30:
How did you create the pcap files?

You could also try this:
[image]

The difference there is `_timestamp` is the received-message timestamp.

Jumps like the 30ms *probably* correspond to scheduling issues.

Can you verify you see the same thing, please?

Note that doing this at 100Hz is relatively new to ArduPilot - there very well be issues.

Eli-Pinchover, 2020-01-30:
> How did you create the pcap files?
>
> You could also try this:
> [image]
>
> The difference there is `_timestamp` is the received-message timestamp.
>
> Jumps like the 30ms _probably_ correspond to scheduling [...]

the mavproxy runs on the raspberry-pi.
I do     "--output add  10.0.0.6:14550" (the IP of the other (windows)  PC)  to the mavproxy.
also:  "long SET_MESSAGE_INTERVAL 33 10000"   (i send this every 1 sec)
The wireshark on the win PC  captures the pcap files.
BTW, in your graph I see 0.010 most of the time- isn't that means 100 ms??? I try to get the message every 10 ms!

peterbarker, 2020-01-30:
On Thu, 30 Jan 2020, Eli-Pinchover wrote:

>       How did you create the pcap files?

> the mavproxy runs on the raspberry-pi.
> I do "--output add 10.0.0.6:14550" (the IP of the other (windows) PC) to the mavproxy.
> also: "long SET_MESSAGE_INTERVAL 33 10000" (i send this every 1 sec)
> The wireshark on the win PC captures the pcap files.

There's going to be rather significant jitter introduced by all of that. 
MAVProxy isn't going to be the best tool here :-)

I do have a branch that can generate pcap files on the stm32 itself.  It's 
in my branches on github.... somewhere....  That will give you much 
tighter timings.

> BTW, in your graph I see 0.010 most of the time- isn't that means 100 ms??? I try to get the message every 10 ms!

That's in seconds - so I don't think so...

Peter

Eli-Pinchover, 2020-01-30:
> On Thu, 30 Jan 2020, Eli-Pinchover wrote: How did you create the pcap files? the mavproxy runs on the raspberry-pi. I do "--output add 10.0.0.6:14550" (the IP of the other (windows) PC) to the [...]

Well, let me put my goal from the first place:

I'd like to position and attitude messages from Pixhawl/Ardupilot in 50 Hz rate (i.e every 20 ms, +- 2 ms)
Is this feasible?
b.t.w: can we talk directly (whatsup,...)??

peterbarker, 2020-01-30:
On Thu, 30 Jan 2020, Eli-Pinchover wrote:

> that can generate pcap files on the stm32 itself. It's in my branches on github.... somewhere.... That will give you much tighter timings.
>       BTW, in your graph I see 0.010 most of the time- [...]

20ms is doable as an *average* rate - I *think* my graphs show that.

+/-2ms is not doable if that's a hard requirement - that would mean any 
time we can't schedule a send in a loop (because we're busy flying the 
vehicle....)  you would violate the requirement (assuming 400Hz).

We don't like having scheduling issues like this - but we prefer our 
vehicles to not glitch in their flying characteristics and perhaps not 
shove telemetry out this loop.  We end up with code like this:

     // minimum amount of time (in microseconds) that must remain in
     // the main scheduler loop before we are allowed to send any
     // mavlink messages.  We want to prioritise the main flight
     // control loop over communications
     virtual uint16_t min_loop_time_remaining_for_message_send_us() const {
         return 200;
     }

Peter

Eli-Pinchover, 2020-01-31:
> On Thu, 30 Jan 2020, Eli-Pinchover wrote: that can generate pcap files on the stm32 itself. It's in my branches on github.... somewhere.... That will give you much tighter timings. BTW, in your [...]

Ok Thanks Peter.
As I wrote, I need to get the position (#33) and attitude (#30) messages at an accurate coherent 50 Hz rate.
Any idea of how to achieve this goal?

wly2014, 2021-02-14:
> [image]
> 
> Works For Me™
> 
> I don't think you can expect this to work on 3.6.12 - pretty sure it was new for 4.0

Hi peterbarker, What software did you use in the figure? thanks.

peterbarker, 2021-02-14:
On Sun, 14 Feb 2021, wly wrote:

>       image
>
>       Works For Me™
>
>       I don't think you can expect this to work on 3.6.12 - pretty sure it was new for 4.0
> 
> Hi peterbarker, What software did you use in the figure? thanks.

I probably used MAVProxy's messagerate module - but QGC and MissionPlanner 
both have mavlink inspectors which will give you a rate, I think.