ArduPilot/ardupilot #13326: Yaw drift too high when not using magnetometer

NitinJSanket, 2020-01-19:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

The Yaw drift on APM much much higher than that on Betaflight/Cleanflight when not using a magnetometer. We observed drift of about 1 degree in 1.5 mins on betaflight when sitting on the desk and in APM 1 degree drift was observed in about 10 secs. Tested this on various autopilots and the performance is the same. Our guess was that this is because of the EKF not tuned correctly when not using a magnetometer, so we tried reducing the noise values of gyro and acc but it didn't really make much of a difference. Any ideas?

**Version**
3.6.11 - 4.0.0

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X

**Hardware type**
Kakute F7 AIO, Kakute F7 V1.5, Pixracer, mRo Pixhawk 

**Logs**
_Please provide a link to any relevant logs that show the issue_

NitinJSanket, 2020-01-19:
Also, I'm operating indoors with no optical flow and no GPS.

NitinJSanket, 2020-01-21:
Ok, Here's what I think. Even if you disable EKF usage for attitude estimation using AHRS_EK2_USE = 0, I think the attitude estimation using DCM uses gyro scale and biases updated from EKF, making the drift crazy. Here's a snapshot and the log accompanied.
[image]

[2 12-31-1969 7-00-00 PM.zip](https://github.com/ArduPilot/ardupilot/files/4093109/2.12-31-1969.7-00-00.PM.zip)

rmackay9, 2020-01-22:
I suspect that the issue is that we calibrate the gyros once soon after startup and perhaps before the board and sensors have warmed up. I'll bet if the gyro calibration is done again later the drift will be much less.  I think it's possible to re-do the gyro calibration from MP's Flight Data, Action screen (see top left drop-down).

NitinJSanket, 2020-01-23:
It did become better, but the drift is still too high, its about 10 degrees in 3 mins now, the betaflight drifts far far less, about 1 degree in 90 secs.