ArduPilot/ardupilot #13335: Change already flying speed condition for takeoff flight mode

urlu75, 2020-01-19:
# Feature request

**Is your feature request related to a problem? Please describe.**
New takeoff mode does not allow to run before hand launching.
Running before launching helps with heavy plane.
 If plane.ahrs.groundspeed() > 3  when accell level is detected, takeoff is skipped and circle mode is engaged resulting in a plane with no speed and not altitude trying to circle.

**Describe the solution you'd like**
use stall speed based on ARSPD_FBW_MIN or use a greater value (6 m/s ?) or let user to choose with a new paramater

**Describe alternatives you've considered**
Continue to use auto flight mode with a mission.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Additional context**
https://discuss.ardupilot.org/t/new-takeoff-mode-skipped-for-circle-just-after-launch/49756

tridge, 2020-03-03:
should check the started_flying_ms and see if we have been flying for at least 5s

peterbarker, 2020-03-03:
@urlu75 did you want to do a PR for this?

urlu75, 2020-03-03:
Thanks !
It seems that this PR [Plane: prevent jump to Circle upon Takeoff mode entry unless flying](https://github.com/ArduPilot/ardupilot/commit/7828274c72597458b1002b09e722d6924f8dbf5d) solve this issue.
I guess this can be closed.

closed 2020-03-03 by urlu75