ArduPilot/ardupilot #13347: ZigZag mode - Allow altitude control

JacksonUAS, 2020-01-21:
### Feature request

Add altitude control via "throttle" stick to ZigZag mode, in a similar manner to how yaw control is currently managed. This would more closely align it to other flight controllers A-B mode etc setups. 

**Platform**
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2020-02-06:
To add a bit more detail, the request is to add altitude control support as the vehicle is transitioning autonomously between points A and B?  I think the pilot can already control the altitude when the vehicle is at the edges.

JacksonUAS, 2020-02-06:
> To add a bit more detail, the request is to add altitude control support as the vehicle is transitioning autonomously between points A and B? I think the pilot can already control the altitude when the vehicle is at the edges.

Correct, in the same way as a DJI agras does.