ArduPilot/ardupilot #13353: LOG_DISARMED = 1 and downloading log files over MAVLINK

DanS87, 2020-01-22:
Download Logfiles over Mavlink while LOG_Disarmed is enabled leads to a lot of new files.
It seems between every File that gets downloaded a new log file is created and then aborted when the next file is read for transmission.

That means if I have 20 Logfiles on my card and I download all of them over Mavlink I end up with 40 Logfiles on the SD Card.
This, of course, gets exponentially worse

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2020-01-22:
On Wed, 22 Jan 2020, DanS87 wrote:

> That means if I have 20 Logfiles on my card and I download all of them over Mavlink I end up with 40 Logfiles on the SD Card.
> This, of course, gets exponentially worse

I think now that NuttX is out of the picture we can look at having that 
second filehandle open.

muramura, 2020-01-24:
I downloaded a 36MB file with a USB cable with the same settings on AC3.6.12, and a lot of files of several kilobytes to several tens of kilobytes were created.
I've experienced that without a download operation, it is not created.

[image]

WickedShell, 2020-01-24:
@peterbarker I haven't done extensive testing with it by any means, but Lua has had second (and maybe third) file handles open with no noticed trauma on ChibiOS boards, including repeatedly opening/reading/writing/closing from files other then the log one.

closed 2021-03-23 by peterbarker