ArduPilot/ardupilot #13355: During terrain following send_position_target_global_int() sends AGL altitude along with MAV_FRAME_GLOBAL_INT frame.

MikeCharikov, 2020-01-22:
### Bug report

When flying auto mode with rangefinder ArduCopter sends AGL altitude in POSITION_TARGET_GLOBAL_INT message instead AMSL (it's ok, because autopilot doesn't know target AMSL) but frame is still MAV_FRAME_GLOBAL_INT (cause it is hardcoded in send_position_target_global_int()). 

` mavlink_msg_position_target_global_int_send(
        chan,
        AP_HAL::millis(), // time_boot_ms
        **MAV_FRAME_GLOBAL_INT, // targets are always global altitude**
        0xFFF8, // ignore everything except the x/y/z components
        target.lat, // latitude as 1e7
        target.lng, // longitude as 1e7
        target.alt * 0.01f, // altitude is sent as a float`

Autopilot should send MAV_FRAME_GLOBAL_TERRAIN_ALT_INT when uses terrain_following.
 

**Version**
Copter-3.6
Copter-4.0.0

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL

rmackay9, 2020-01-23:
Mike, this should go into the support forums first before being raised here I think.

Is it possible that EK2_SLAT_SOURCE = 1 (for rangefinder)?  If "yes" then this isn't the correct way to do terrain following.  The setup is here on the wiki: https://ardupilot.org/copter/docs/terrain-following.html#terrain-following

rmackay9, 2020-01-23:
I'm going to close this issue but we can re-open it if it turns out to be a real issue.  I'll monitor for replies here as well.

closed 2020-01-23 by rmackay9

MikeCharikov, 2020-01-23:
> Is it possible that EK2_ALT_SOURCE = 1 (for rangefinder)? If "yes" then this isn't the correct way to do terrain following. The setup is here on the wiki: https://ardupilot.org/copter/docs/terrain-following.html#terrain-following

Thanks Randy. No, our setup is correct and we already performed real flights with terrain following. 

Ok, i'll move this topic to forum, but just for history I'll clarify here our problem: Our mission can include WPs in both relative and terrain frames. We use this telemetry field for displaying target point. But when copter if flying towards "terrain" WP - altitude value in POSITION_TARGET_GLOBAL_INT telemetry message is inconsistent with frame. Additionally there is no way now to clearly understand via telemetry what is the frame of current WP.

So, it looks like the bug for me because of inconsistency between alt and frame values.

peterbarker, 2020-01-23:
On Wed, 22 Jan 2020, MikeCharikov wrote:

> Is it possible that EK2_ALT_SOURCE = 1 (for rangefinder)? If "yes" then this isn't the correct way to do terrain following. The setup is here
>       on the wiki: [...]

The altitude we're supplying in POSITION_TARGET_GLOBAL_INT isn't in global 
frame when we're giving that as the frame?  That sounds bad.  Can you 
provide a log showing that, please?

There's an assumption here: 
https://github.com/ardupilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L103

that the waypoint returned by get_wp is in the global frame.  That's where 
I'll be looking first...

Not returning the altitude in TERRAIN altitude if we're flying terrain 
isn't great, but more lack-of-feature rather than a bug.  Returning 
something in terrain alt in global-position-int can lead to problems for 
the GCS - if it wants to know your AMSL then it would need to have the 
same concept of terrain as you do.

MikeCharikov, 2020-01-23:
Ok, i rechecked it. I'm very sorry but i was wrong. 

POSITION_TARGET_GLOBAL_INT sends AMSL altitude. It is calculated as Origin AMSL+ Waypoint altitude. It has a little of use during terrain following, but it is defently not a bug.  

Yesterday for some reason i made my test on SITL with Origin AMSL = 0 (Oh, I swear i never do that usually!). And therefore i was confused with the result.

Sorry again.