ArduPilot/ardupilot #13357: No visualizations display in SITL

erinline, 2020-01-22:
When I run the command:
`../Tools/autotest/sim_vehicle.py --map --console`
it seems to run SITL simulation, but no visualizations appear. No map, no console, and no MAVProxy command prompt. I do see these errors:
`Failed to load module: No module named 'console'. Use 'set moddebug 3' in the MAVProxy console to enable traceback
Failed to load module: No module named 'map'. Use 'set moddebug 3' in the MAVProxy console to enable traceback
`
However I cannot implement the suggested fix because I the MAVProxy console has not been launched. I also can't do anything with the SITL simulation without the console. How do I fix this?

#### Platform
SITL/copter.

khancyr, 2020-01-22:
Hello, 

It is probable that you miss some mavproxy depencies likes python-wxgtk3.0 . Please check you install mavproxy correctly from http://ardupilot.github.io/MAVProxy/html/index.html.

About the console, the default terminal you get with SITL is mavproxy ! That is where you read the error.

peterbarker, 2020-01-23:
On Wed, 22 Jan 2020, erinline wrote:

> When I run the command:
> ../Tools/autotest/sim_vehicle.py --map --console
> it seems to run SITL simulation, but no visualizations appear. No map, no console, and no MAVProxy command prompt. I do see [...]

If @khancyr 's sugestions don't work, try

set moddebug 4
module load map

That should give you more diagnostics to play with.

erinline, 2020-01-23:
Ah I see, thanks @khancyr for explaining the mavproxy terminal! Regarding missing dependencies, I did get some errors when installing mavproxy. Specifically:
```
Package python3-wxgtk3.0 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package python3-pygame is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package python3-opencv
E: Package 'python3-wxgtk3.0' has no installation candidate
E: Package 'python3-pygame' has no installation candidate
```

And when I tried @peterbarker 's suggestion (but with moddebug 3 bc 4 was rejected) the source of the error seems to be:
```
import cv2
ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type
```

However I'm not sure where to go from here, as I do have opencv installed, so I don't know why it can't load.

erinline, 2020-01-23:
Fixed: removed all pip installed files (i.e. removed /.local), did pip install MAVProxy, and it works just fine. Shrug.

closed 2020-01-23 by erinline

nefelimet, 2021-01-04:
@erinline you mean that you completely deleted the .local folder?

dpd97, 2021-02-01:
unrelated question, I want to be able to post issues like this ^^ , when I clicked 'new issue' it asked me if I had a proposal or a bug to report..... if I simply want to ask about somewhere I am stuck (like in this post) which do I use? Or is there another way of posting an issue? 

Thanks!

auturgy, 2021-02-01:
> unrelated question, I want to be able to post issues like this ^^ , when I clicked 'new issue' it asked me if I had a proposal or a bug to report..... if I simply want to ask about something I am stuck on (like in this post) which do I use? Or is there another way of posting an issue?

User support Is via the for at discuss.ardupilot.org
Developer support is on discord https://ardupilot.org/discord
We aim to keep the GitHub issues list for confirmed bugs and feature requests - gets too noisy and hard to manage otherwise.

Zarqu0n, 2021-12-20:
İf you using anaconda, download Mavproxy requirements with conda.
E.g conda install -c anaconda wxpython 
This solved for me.

furkanunl11, 2023-09-14:
Thats work for me Zarqu0n. Thanks!

> Anaconda kullanıyorsanız, Mavproxy gereksinimlerini conda ile indirin. Örneğin conda install -c anaconda wxpython Bu benim için çözüldü.

sukuna2601, 2024-06-07:
try "pip install opencv-python" it worked for me