ArduPilot/ardupilot #13358: Two Same type Rangefinder Error when use 2 differents Serial Lidar together

framosnt, 2020-01-22:
### Bug report

**Issue details**

Hi team,

My project is to Pixhawk 1 - Arducopter 4.0.1 with 2 radars type UART, but each serial port are different type ! first radar is 1 sector to altimeter and secund is 3 sector to proximity.

The each radar work very well when is alone, but together have a conflit.

First Radar (have a 1 sector) to altimeter:
Serial2_type = 9 (Rangefinder)
RNG_Orient= 25 (down)
RNG_type = 11 (uLanding)

Secund Radar (have 3 sectors) to proximity:
Serial4_type = 11 (Lidar360)
RNG_Orient= 0 (Forward)
RNG_type = 11 (uLanding)
apx_type = 3 (TeraRangerTower)

When I try this, the RangeFinder1 show the correct value, when proximity is not detect, alternating this.
**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

JacksonUAS, 2020-01-24:
Why are you setting rng settings for the proximity sensors?

framosnt, 2020-01-24:
> Why are you setting rng settings for the proximity sensors?

Dear Jackson, I use two radar, to follow terrain (RNGFND1_ORIENT = 25)  not necessary to setting as proximity sensor, but other to avoidance ( (RNGFND1_ORIENT = 0) is 3 sector radar and need to show detected object in Proximity Window !

The problem is in function FIND_SERIAL, the parameter "procotol" is ignored, I have two serial radar with different protocol, one is ONE SECTOR is RANGEFINDER serial Protocol other is 3 SECTOR, serial Protocol is Lidar360. When I setting Proximity Type TerarangerTower, the function FIND_SERIAL would return only Lidar360 serial port !!!!

When I use only one radar typed RANGEFINDER worked well and show RANGEFINDER VALUE, but wrote in HUD error menseger "LIDAR ERROR", because not identify that my radar use uLANDING RANGEFINDER is SINGLE POINT and can to be a ORIENTATION to down side !!!! If I change Orient to 0 (forward) the error not show !

The code have a incorrect TYPE RANGEFINDER identification !

I did test, created new RANGEFINDER , change the code of Rangerfinder LEDDARONE to same code of ULANDING, did setting first Rangefinder to uLanding and other to LeddarOne, same this change type of Rangefinder this returned first Serial AVAILABLE  !!!

I will upload log and setting files !

framosnt, 2020-01-24:
[image]

framosnt, 2020-01-24:
SYSID_SW_MREV	120
SYSID_THISMAV	1
SYSID_MYGCS	255
PILOT_THR_FILT	0
PILOT_TKOFF_ALT	0
PILOT_THR_BHV	0
SERIAL0_BAUD	115
SERIAL0_PROTOCOL	2
SERIAL1_PROTOCOL	1
SERIAL1_BAUD	115
SERIAL2_PROTOCOL	11
SERIAL2_BAUD	115
SERIAL3_PROTOCOL	5
SERIAL3_BAUD	38
SERIAL4_PROTOCOL	9
SERIAL4_BAUD	115
SERIAL5_PROTOCOL	-1
SERIAL5_BAUD	57
SERIAL1_OPTIONS	0
SERIAL2_OPTIONS	0
SERIAL3_OPTIONS	0
SERIAL4_OPTIONS	0
SERIAL5_OPTIONS	0
SERIAL_PASS1	0
SERIAL_PASS2	-1
SERIAL_PASSTIMO	15
TELEM_DELAY	0
GCS_PID_MASK	0
RTL_ALT	1500
RTL_CONE_SLOPE	3
RTL_SPEED	0
RTL_ALT_FINAL	0
RTL_CLIMB_MIN	0
RTL_LOIT_TIME	5000
RNGFND_GAIN	0.8
FS_GCS_ENABLE	0
GPS_HDOP_GOOD	140
SUPER_SIMPLE	0
WP_YAW_BEHAVIOR	2
LAND_SPEED	50
LAND_SPEED_HIGH	0
PILOT_SPEED_UP	250
PILOT_ACCEL_Z	250
FS_THR_ENABLE	1
FS_THR_VALUE	975
THR_DZ	100
FLTMODE1	0
FLTMODE2	0
FLTMODE3	0
FLTMODE4	0
FLTMODE5	0
FLTMODE6	0
FLTMODE_CH	5
SIMPLE	0
LOG_BITMASK	176126
ESC_CALIBRATION	0
TUNE	0
FRAME_TYPE	1
ARMING_ACCTHRESH	0.75
ARMING_RUDDER	2
ARMING_MIS_ITEMS	0
ARMING_CHECK	1
DISARM_DELAY	10
ANGLE_MAX	4500
PHLD_BRAKE_RATE	8
PHLD_BRAKE_ANGLE	3000
LAND_REPOSITION	1
FS_EKF_ACTION	1
FS_EKF_THRESH	0.8
FS_CRASH_CHECK	1
[... 921 lines cut ...]
TCAL_ENABLED	0
SRTL_ACCURACY	2
SRTL_POINTS	300
PILOT_SPEED_DN	0
LAND_ALT_LOW	1000
FHLD_XY_P	0.2
FHLD_XY_I	0.3
FHLD_XY_IMAX	3000
FHLD_XY_FILT_HZ	5
FHLD_FLOW_MAX	0.6
FHLD_FILT_HZ	5
FHLD_QUAL_MIN	10
FHLD_BRAKE_RATE	8
FOLL_ENABLE	0
AUTOTUNE_AXES	7
AUTOTUNE_AGGR	0.1
AUTOTUNE_MIN_D	0.001
SCR_ENABLE	0
TUNE_MIN	0
TUNE_MAX	0
OA_TYPE	0
SID_AXIS	0
FS_VIBE_ENABLE	1
FS_OPTIONS	0
CAM_RC_TYPE	0

framosnt, 2020-01-24:
[image]

only one radar was working, but with wrong information !

framosnt, 2020-01-24:
[image]

framosnt, 2020-01-24:
[2020-01-20 10-49-49.rar.zip](https://github.com/ArduPilot/ardupilot/files/4108882/2020-01-20.10-49-49.rar.zip)
obs.: change to .rar file to extract !

framosnt, 2020-01-24:
[HEXACOPTER WITH ONE RADAR RANGEFINDER ONE SECTOR.mp4.ZIP](https://github.com/ArduPilot/ardupilot/files/4108907/HEXACOPTER.WITH.ONE.RADAR.RANGEFINDER.ONE.SECTOR.mp4.ZIP)
OBS> VIDEO, only change the extension to MP4 !!!

rmackay9, 2020-01-27:
Thanks for the report.

Normally it's best to report support issues first in the forums: https://discuss.ardupilot.org/c/arducopter/copter-40.

IamPete1, 2021-09-12:
sorry we never got to this.

closed 2021-09-12 by IamPete1