ArduPilot/ardupilot #13365: AC_Avoidance/AP_Proximity: use bearing instead of heading for database importance

RoelHelsen, 2020-01-23:
## Feature request
### Issue
I'm now experimenting with the obstacle avoidance and upon reading the code for the obstacle avoidance database combined with the SF40 publishing obstacles, I found that the vehicle's heading  (the direction, the front of the vehicle is facing) is being used to determine the importance of the points.
This makes sense for a Rover, for which the library was originally developed, but for Copter the front of the vehicle is not always pointing in front of the movement.

### Proposed change
Therefore I think it would be better to use the vehicle's bearing (the direction in which the vehicle is moving) for determining the importance of the database points. This is most advantageous for Copter but can benefit the other vehicle types as well. Airplanes can move a bit sideways due to wind and also for Rovers with omni-wheels(if these are supported) this might be more appropriate

IamPete1, 2021-09-12:
@RoelHelsen Maybe you can point out the offending code, I cant see what we are doing wrong after a quick look. It is possible that this has been fixed by @rishabsingh3003

rishabsingh3003, 2021-09-12:
@IamPete1 I am not aware of any such logic inside the OA DB. When we push an obstacle inside the DB, we adjust the radius in proportion to the distance to the obstacle. So obstacles far away are perceived as circles (now sphere in 4.1) with a larger radius, and vice - versa for closer objects. After that, every obstacle is treated the same inside Bendy Ruler... i.e a sphere.

rishabsingh3003, 2021-09-12:
Maybe @rmackay9 knows what this is about

rmackay9, 2021-09-13:
Tom Pittenger wrong the first iteration of the OA DB and had some code in there to try infer the importance of obstacles based upon where they were relative to the vehicle's heading... I removed that complexity some time ago so we can close this.  txs

closed 2021-09-13 by rmackay9

magicrub, 2021-09-13:
That "complexity" made it awesomer.