ArduPilot/ardupilot #13374: IOMCU freezes periodically after upgrading Copter 3.6.9 to 3.6.11 or 3.6.12

RickReeser, 2020-01-24:
### Bug report

Copter 3.6.11
Cube Black

**Issue details**
Sometimes while flying, the drone will briefly lose control and then recover. The logs show that this is caused by the motor outputs freezing. This appears to be an issue with the IOMCU, as VServo also freezes.

[image]

I have encountered this issue on several different drones, although reproduction of this problem is hit-and-miss. These are the steps I did to encounter it:

1. Use Mission Planner to upgrade Cube Black firmware from custom 3.6.9 to custom 3.6.12 (details on custom firmware below)
2. After upgrading, connect MP. There's an error **"Check BRD_TYPE: Failed to update IO firmware."** This error appears even with official firmwares.
3. Power cycle the board by unplugging USB and plugging it back in. Connect MP. The error is gone.
4. Fly the drone. The FMU freezing issue may or may not appear.

At this point, to fix this, I have tried:
- Changing firmware to official 3.6.11, 3.6.12. Problem still present.
- Change firmware back to custom 3.6.9 or official 3.6.9. Not sure if problem is present (see below)

After going back to 3.6.9, I found some events that look like this:
[image]

They're very short (only 1 log interval between them) and VServo may or may not be frozen, depending on log interval aliasing.

At first, I suspected hardware as the culprit, but I have encountered this on multiple Cubes. So I then suspected the IOMCU firmware update in Copter 3.6.10. This is possible, but depends on whether the events in my 3.6.9 are real RCOut freezes, or just logging artifacts.

The custom firmware I mentioned are:
- 3.6.9 custom has some small changes to hardcoded things like disabling the rangefinder prearm check and ports in the precision landing behavior from 3.5.7. I have been flying this for a long time with no issues.
- 3.6.11 custom has the same, plus some backported commits from master, maybe the most relevant being the stm32 peripheral power bootup delay: https://github.com/ArduPilot/ardupilot/pull/11319/files 
- In both, I have a few custom parameters. I haven't yet tried wiping EEPROM.

**Logs**
Logs included are 2 short flights on 3.6.11 custom and official which show 1 event each, plus 2 logs on 3.6.9 official with the possible log alias events.
[fmu freezing.zip](https://github.com/ArduPilot/ardupilot/files/4111207/fmu.freezing.zip)

rmackay9, 2020-01-25:
@RickReeser,

So just to clarify this issue happens with the official software as well?  I've been chasing this ESC Sync issue which could be related: [discussion](https://discuss.ardupilot.org/t/esc-sync-problem-with-update-from-3-6-9-to-3-6-10/48261), [issue](https://github.com/ArduPilot/ardupilot/issues/13310).

RickReeser, 2020-01-25:
Correct.
That said, I've only tried official firmware after first loading my custom 3.6.11 firmware. I haven't tried going from 3.6.9 directly to official 3.6.11, in case that matters.

auturgy, 2020-01-25:
See https://github.com/ArduPilot/ardupilot/pull/12033

rmackay9, 2020-01-25:
@auturgy, thanks for the link.  I've checked the Copter-3.6 branch and it looks like this fix went in between 3.6.10 and 3.6.11 so I think that disqualifies it as either the cause or the fix for this issue.

The big change we made between Copter-3.6.9 and 3.6.10 was moving to the ChibiOS IO firmware.

rmackay9, 2020-01-28:
@RickReeser,

We discussed this on the dev call and it seems that Copter-4.0.x has a new IOMC onboard log message that will tell us is the IO processor has restarted in flight.  The IO processor restarting could lead to the type of output freeze you're seeing.

Do you think you could re-test with Copter-4.0.1 and provide a log?

Thanks again for this report

tridge, 2020-01-28:
the IOMC will also give us information on other types of errors in the comms protocols between FMU and IOMCU

RickReeser, 2020-01-29:
Will do. It seems to come and go, but I'll start flying 4.0 and save the log if it happens again.

rmackay9, 2020-02-01:
We think this is resolved with this PR which is included in Copter-4.0.2-rc1 ([release announcement](https://discuss.ardupilot.org/t/copter-4-0-2-rc1-available-for-beta-testing/51790)) so I'm going to close this.  Thanks again very much for the report!

closed 2020-02-01 by rmackay9