ArduPilot/ardupilot #13379: EKF is consuming rangefinder altitudes when rangefinder status is not Good

rmackay9, 2020-01-25:
This issues stems from this discussion in the Copter-4.0 category: https://discuss.ardupilot.org/t/lidar-in-althold/51090

The user's vehicle is setup with EK2_ALT_SOURCE = 1 (RangeFinder) because it is being used in an indoor environment where the air conditioning system is upsetting the barometer.  The user is not attempting to do [terrain following which is documented here](https://ardupilot.org/copter/docs/terrain-following-manual-modes.html) and [here](https://ardupilot.org/copter/docs/terrain-following.html) and would NOT involving modifying EKF parameters.

The user's log clearly shows the EKF is consuming the rangefinder's altitude even when it's status is 2 (2="OutOfRangeLow", scale is on right).  The EKF's altitude is shown in red.
[image]

priseborough, 2020-01-27:
Can you please provide a link to where that message is logged in code. When I check the code I can see a valid measurement count where a value of 2 does not necessarily mean it is invalid. The EKF checks published range against the upper and lower limit and in the past an invalid range was indicated by an out of range reading.

rmackay9, 2020-01-27:
@priseborough, an invalid value could also be too short though in case that's important.

[This is where the rangefinder library publishes the RFND log message](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder.cpp#L668).

priseborough, 2020-01-27:
Looks like the EKF is checking that data item: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF2/AP_NavEKF2_Measurements.cpp#L54

Given that is the case, I would start checking the vertical position innovations as they will show what data has made it through the EKF checks.

priseborough, 2020-01-27:
Something to be aware of is that when operating without vertical velocity observations from GPS, the EKF is significantly more sensitive to errors or dropouts in height sensor fusion.

rmackay9, 2020-01-27:
@priseborough, ok, so you think it's all working as expected?  So the changes in the EKF's estimated altitude (shown in the graph above which comes from the log from the discussion) is just because the EKF doesn't have a good source of altitude information?

anbello, 2020-01-28:
> drift because it relies on IMU only

If this is the case, why it relies on IMU only instead of IMU + baro? 
Sorry for silly question but I am not able to understand this from the code.

peterbarker, 2020-11-25:
I believe this was fixed by https://github.com/ArduPilot/ardupilot/pull/14375

closed 2020-11-25 by peterbarker