ArduPilot/ardupilot #13390: Arming Quadcopter and launching without GPS and Compass with high stability using built in Inertial sensor data and Barometer data.

gauthampkrishnan, 2020-01-27:
### Feature request

**Is your feature request related to a problem? Please describe.**
The problem that I found in the Ardupilot software is that user cant ARM or launch the quad copter without buying an extra GPS or Compass module which adds to the extra cost other than the Pixhawk Module. Other cheap flight controllers are able to ARM and stabilize the quad copter without the help of the GPS.

**Describe the solution you'd like**
The solution I like to give is to collect data from built in IMU and barometer and try to stabilize the quadcopter using the above listed sensors.

**Describe alternatives you've considered**
The alternative solution i have considered is to collect data from IMU sensors and barometric pressure sensors and do the auto PID tuning to make the quad copter to stabilize.

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
No screen shots are available as the feature is not available in Ardupilot.

rmackay9, 2020-01-27:
It's should be possible to arm and takeoff without a GPS in acro, stabilize or AltHold modes.  The only thing you should need to be careful about though is that no geofence has been defined.  If a geo-fence is defined then a pre-arm check will stop you from arming.

I'm going to close this but we can re-open if necessary.

By the way, it's best to report [issues in the forums first](https://discuss.ardupilot.org/c/arducopter/copter-40) so that we can keep this issues list for confirmed bugs and enhancement requests.  No big deal of course, we appreciate the feedback in any forum but if possible the support forums are a good place to start.

closed 2020-01-27 by rmackay9

lthall, 2020-01-27:
Just to be sure I checked that nothing has changed that would prevent arming without gps or external compass using just a receiver and 3dr pixhawk. I can confirm that we can still arm in manual modes without any external sensors.