ArduPilot/ardupilot #13396: UAVCAN ESCs: send ESC RPM to GCS using ESC_TELEMETRY_xxx message

rmackay9, 2020-01-28:
It seems that UAVCAN does not send RPM (and other data) to the ground station in real-time using the ESC_TELEMETRY_xxx_to_yyy messages unlike other advanced ESCs including BLHeli, KDECAN, PiccoloCAN and ToshibaCAN.

[GCS_Common.cpp's try_send_message probably needs to be enhanced around here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L4560) although it would probably also make sense to move all this logic which handles the protocols into a separate function called send_esc_telemetry()

gauthampkrishnan, 2020-01-30:
Which logic should be moved into the send_esc_telemetry()? Did you meant the if conditions?

TunaLobster, 2021-02-02:
This was completed in #14851.

peterbarker, 2021-02-02:
Thanks @TunaLobster (and @elhayra :-) )

closed 2021-02-02 by peterbarker