ArduPilot/ardupilot #13407: mavlink message HIL_ACTUATOR_CONTROLS not sent

richardor-cell, 2020-01-28:
#### Issue details
Hi, creating an octaquad SITL with gazebo_ardupilot using libgazebo_motor_model consuming HIL_ACTUATOR_CONTROLS mav msg but it sems the msg is not yet available. Only .h file exist but not .cpp

#### Version
v4.0.1 dev

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
X8

#### Hardware type
SITL, HIL
#### Logs
no logs

richardor-cell, 2020-01-29:
Errata, it is gazebo_mavlink_interface plugin who consumes hil_actuator_controls msg.

peterbarker, 2024-07-31:
I think our Gazebo interface has evolved, so thi is no longer relevant.

Pity we didn't get to this sooner than now...

closed 2024-07-31 by peterbarker