ArduPilot/ardupilot #13411: AP_RangeFinder: some noise in TFMiniPlus-I2C

tatsuy, 2020-01-29:
**Issue details**
I placed TFMiniPlus-I2C at the same place and measured the distance for 3 minutes.
There are a lot of noise in the graph.
[image]

I have some TFMiniPlus. And this issue happens in all devices in I2C.
But it doesn't happen using serial mode.

**Version**
ArduCopter V4.0.2-dev (56044b8b)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Logs**
[00000182.zip](https://github.com/ArduPilot/ardupilot/files/4125984/00000182.zip)

lukezhqin, 2020-02-02:
I indeed found this i2c issue and I chose to use serial port then ...I thought that i2c is not that reliable as we hope.

muramura, 2020-02-03:
@tatsuy san.
This driver does the checksum calculation. The data is good and the quality of the distance is judged. I feel this quality judgment is loose. I think it would be easier to consider logging this quality value.

peterbarker, 2020-02-03:
@muramura that sounds like a good idea.  Do you feel like knocking a patch together to put it into `log_RFND`?

Presumably it will go in as NaN if none is available.

tatsuy, 2020-02-06:
Do you mean strength value of TFmini?
I logged it.
[00000062_tfminiplus-i2c_noise.zip](https://github.com/ArduPilot/ardupilot/files/4162840/00000062_tfminiplus-i2c_noise.zip)

Strength and distance seem to have little correlation.
[image]

wood1139, 2020-03-02:
The "i2c read" operation will interrupt the timing sequence in TFMiniPlus. Keep the i2c reading frequency lower than the sample rate will solve this problem. When TFMiniPlus is working under 100Hz sample rate, a practice i2c reading frecuency is 40Hz.

lukezhqin, 2020-03-02:
> The "i2c read" operation will interrupt the timing sequence in TFMiniPlus. Keep the i2c reading frequency lower than the sample rate will solve this problem. When TFMiniPlus is working under 100Hz sample rate, a practice i2c reading frecuency is 40Hz.

@wood1139 ,Hi, is there any proof for this statement?

wood1139, 2020-03-03:
> @wood1139 ,Hi, is there any proof for this statement?

I got this from Benewake F.A.E. and it really works.

lukezhqin, 2020-03-03:
> I got this from Benewake F.A.E. and it really works.

@wood1139 thanks

tatsuy, 2020-03-09:
@wood1139 Thank you!
I confirmed that it works.
I tested using 40Hz, 33Hz and 20Hz. and I've decided to use 33Hz.

**40Hz**
[image]
There are still a few noise.

**33Hz**
[image]

**20Hz**
[image]

PR #16102 (2020-12-19): AP_RangeFinder: TFminiPlus: change the frequency setting
This is the replacement of #13765 and #13300. and this can fix #13411 and #12396.

Benewake says we can set TFminiPlus' frame rate to 250Hz here.
https://github.com/TFmini/TFminiPlus-IIC-Pixhawk/blob/master/Application%20of%20TFmini%20Plus%20IIC%20communication%20in%20Pixhawk.pdf
[image]
So I changed the frame rate from 100Hz to 250Hz.

I also include #13765 and #13300 (a bit modified) for this PR.
I tested using some combination of these rates.
TFmini Plus frame rate: 100Hz, ArduPilot TFmini Plus driver's rate: 100Hz (current master)
TF [... cut]
closed 2020-12-19 by lucasdemarchi