ArduPilot/ardupilot #13412: SITL does not start with DJIX and some other frame types

rishabsingh3003, 2020-01-29:
### Bug report

**Issue details**

When I enter the following command - 
` sim_vehicle.py -v ArduCopter -f djix  --map --console`
It gives the following error:

[image]

Neither can I find the "default_params/copter-djix.parm" anywhere. Is this a bug? Forgive me if I am doing something wrong. I have noticed a similar issue for a few other frames. 

**Version**
Master

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

auturgy, 2020-01-29:
It just means that no one has pushed a parameter file for those frame types.
You can either start sitl with defaults and then set your own parameters, or generate a parameter file and pr it.

Regards,

James

IamPete1, 2022-08-05:
Closed by https://github.com/ArduPilot/ardupilot/pull/19010

closed 2022-08-05 by IamPete1