ArduPilot/ardupilot #13414: DataFlash-over-MAVLink broken by multiple links

peterbarker, 2020-01-29:
### Bug report

**Issue details**

If ArduPilot sees traffic on multiple links then Dataflash-over-MAVLink can fail to start at all.

MAVProxy's dataflash_logger module sends a targetted at comonent id 155 and system id 0 - that is, requests any system to provide data from a logging source.

If no route is found to any other system then we handle the packet locally, and start logging just fine.  This will happen if there is only one telemetry link, or at startup when we may not have received a heartbeat from other links.

If we *do* find links to other systems then we forward the packet - and do NOT process the packet locally.  My PR https://github.com/ArduPilot/ardupilot/pull/10527 actually addressed this problem by special-casing `MAVLINK_COMP_ID_LOG` as being a component-ID we control.

I'm not sure we shouldn't just quit using the special component ID for sending these messages.  It causes issues on sending with bodgy mavlink sequence ordering hacks, and it was only put in because mavlink-routing hadn't been invented at the time.

**Version**
master

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
n/a

**Hardware type**
SITL will do

eminakgun, 2020-06-25:
Today I observed strange behaviour in dataflash logs. I have multiple UAVs operating in the field and individual UAV's flashlog logs show the STATUS_TEXT messages of other UAV. How is this possible and is this related to same issue described here?

UAV does not seem to process commands issued by other UAV's comapanion computer like MAV_CMD_DO_CHANGE_SPEED  but shows the status text messages to reply that message like "Set airspeed XX m/s".

I am running ArduPlane 3.9.11, companion computer issues messages with system_id same as the autopilot it is connected to and component_id as 191.

peterbarker, 2020-11-25:
This was closed by https://github.com/ArduPilot/ardupilot/pull/14620

@eminakgun we log any statustext we see... you can use this from a GCS to add notes to the onboard log....

closed 2020-11-25 by peterbarker