ArduPilot/ardupilot #13415: Make error more explicit on install script

khancyr, 2020-01-29:
### Feature request

**Is your feature request related to a problem? Please describe.**
We got some report of use that have failure to use SITL after using the install script.
I am pretty confident that they are just typing the command from the wiki and don't read output and miss the failure

**Describe the solution you'd like**
I propose that we make the error on the install script very visual with a big banner and if possible color !

gauthampkrishnan, 2020-01-30:
Ok So how can we put a banner using html script?

TunaLobster, 2020-05-16:
This is a little stale, but could you post a link to one of the reports? A simple GUI message box might be enough to start off with depending on where the error happens.

zacchaeusliang, 2020-09-22:
Is it in the sim_vehicle.py?

zacchaeusliang, 2020-09-23:
Can you assign this to me?

Hwurzburg, 2020-09-24:
This does not have enough information to take any action...what failure?at what step? there needs to be explicit information on what the issue really is...

khancyr, 2020-09-25:
I got no idea why I raise that ... But I think I want to make explicit that the install script got an issue. Currently it will failed and display a bunch of information but most people don't even read them ... and complain that trying to build is failing. So the goal would be to report the failure on the install script more obviously with something like : 
```
#########################
SOMETHING WENT WRONG !
Task X failed
#########################
```

zacchaeusliang, 2020-09-25:
is this for the STIL install or the general building of the application? what file do you want the errors to be more clear?

void-iish, 2023-07-25:
Could you please assign me the issue ?

Hwurzburg, 2023-07-25:
done

ShankhanilSaha, 2026-01-13:
Hi @Hwurzburg @khancyr , is this Issue still relevant? Its been a while since it was opened and also It has had no activity since Jul 2023. 
If it is, I will look into it and try to write a patch.