ArduPilot/ardupilot #13418: Dataflash over mavlink choppy and incomplete

Pedals2Paddles, 2020-01-29:
### Bug report

In both master and Copter 4.0.1, dataflash over mavlink has become very choppy and incomplete.  See the attached example from master today.  If I download the log from the FC's SD card, the log is perfect and not all choppy.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
3DR Solo

**Hardware type**
Pixhawk 2 cube

**Logs**
[solo log.zip](https://github.com/ArduPilot/ardupilot/files/4131297/solo.log.zip)

peterbarker, 2020-01-30:
On Wed, 29 Jan 2020, Matt wrote:

> In both master and Copter 4.0.1, dataflash over mavlink has become very choppy and incomplete. See the attached example from master today. If I download
> the log from the FC's SD card, the log is perfect and not all choppy.

Any chance of a bisect on this one?

Peter

Pedals2Paddles, 2020-01-30:
Sure!  What's a bisect? :)

peterbarker, 2020-01-30:
On Thu, 30 Jan 2020, Matt wrote:

> Sure! What's a bisect? :)

Well... if you know a commit that works, and one that doesn't, git will 
help you bisect to find the problem - basically you take the midpoint and 
test it.  If it is bad then you move towards the "good" end, if it is good 
you move towards the bad end - where "move" means take the commit halfway 
between this commit and the bad commit.  Repeat until you find what killed 
it.

git bisect start
git bisect bad <BADCOMMITHASH>
git bisect good <GOODCOMMITHASH>

Then, repeatedly:

git submodule update --recursive, config, build, flash-to-solo, test, ...
git bisect good|bad|skip # use skip if the commit is untestable e.g. does 
not build

If there's 1024 commits you will need to do that last bit ~10 times.

Once that's complete, use "git bisect reset" to reset your state.

Peter

Pedals2Paddles, 2020-01-31:
OK. Before I do that, I'm going to try undoing some recent vehicle specific settings to see if it's just a CPU loading issue.

amilcarlucas, 2020-09-10:
Any progress on this one?

xfacta, 2021-01-28:
**Adding to this rather than opening a new issue:**
Checked pixhawks and CUAV Nano v5
Same SD card, same old log files (not even including batch logging for FFT!)
Win10/MissionPlanner (latest stable and also BETA updates)
Linux/QGC

Plane 4.0.8 = cant get a list of logs and cant download
Copter 4.0.5 , 4.0.6 = cant get a list of logs and cant download
Rover 4.0.0 = instantly lists logs, downloads as expected

mavftp works Ok for downloading small logs, but larger logs (eg: including batch logging for FFT) and longer flights produce a CRC error at the end of the download.

I'll try with earlier versions of copter and see if I can find where this broke.

MissionPlanner gives this error trying to obtain a list of logs:

```
Getting list of log files...
Error:System.TimeoutException: Timeout on read - GetLogEntry
   at MissionPlanner.MAVLinkInterface.<GetLogEntry>d__257.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at MissionPlanner.Utilities.Extensions.<>c__DisplayClass1_0`1.<<AwaitSync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at MissionPlanner.Utilities.Extensions.AwaitSync[T](Task`1 infunc)
   at MissionPlanner.MAVLinkInterface.GetLogList()
   at MissionPlanner.Log.LogDownloadMavLink.<LoadLogList>b__12_0()
```

xfacta, 2021-01-28:
I just tried a few Copter firmware versions:

**GOOD** can list and download dataflash logs
============================================
Pixhawk1-1M 00270031 3133470C 36393133
ChibiOS: 1ed2b2e9
ArduCopter V4.0.0 (49693540) 

Pixhawk1-1M 00270031 3133470C 36393133
ChibiOS: cb42c858
ArduCopter V4.0.1 (61f191de)

**BAD** can't list or download dataflash logs
============================================
Pixhawk1-1M 00270031 3133470C 36393133
ChibiOS: cb42c858
ArduCopter V4.0.2 (aae944a8)