ArduPilot/ardupilot #13419: AP_RangeFinder: Benewake TF02 driver should default to use correct I2C address

rmackay9, 2020-01-30:
As discussed on this wiki PR: https://github.com/ArduPilot/ardupilot_wiki/pull/2487 users of the Benewake TFMini I2C lidar must specifically set the RNGFNDx_ADDR parameter to 16.  We should avoid the user having to specifically set this parameter by instead defaulting this parameter to "16" if this sensor is used,

Something similar has already been done in the LeddarVu8 driver.

IamPete1, 2021-09-06:
This is the code that should be copied across from the Leddar backend.

https://github.com/ArduPilot/ardupilot/blob/0ee8b452ab76306d83c6252bcb9dc458bd7380bb/libraries/AP_RangeFinder/AP_RangeFinder_LeddarVu8.cpp#L90-L96

rmackay9, 2021-11-29:
i think we can close this now, thanks @shiv-tyagi !

closed 2021-11-29 by rmackay9