ArduPilot/ardupilot #13439: Rover 4.0 uses EKF3 as a filter, the GCS connection rate is only 10%, and there is no problem using EKF2

chenjiawei, 2020-01-31:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

chenjiawei, 2020-01-31:
Rover 4.0 uses EKF3 as a filter, the GCS connection rate is only 10%, and there is no problem using EKF2

chenjiawei, 2020-01-31:
void GCS_MAVLINK::send_ahrs2()
{
#if AP_AHRS_NAVEKF_AVAILABLE
    const AP_AHRS &ahrs = AP::ahrs();
    Vector3f euler;
    struct Location loc {};
    if (ahrs.get_secondary_attitude(euler) ||  ahrs.get_secondary_position(loc))
    {
        mavlink_msg_ahrs2_send(chan,
                               euler.x,
                               euler.y,
                               euler.z,
                               loc.alt*1.0e-2f,
                               loc.lat,
                               loc.lng);
    }
#endif
}

void GCS_MAVLINK::send_ahrs3()
{
#if AP_AHRS_NAVEKF_AVAILABLE
    const NavEKF2 &ekf2 = AP::ahrs_navekf().get_NavEKF2_const();
    if (ekf2.activeCores() > 0 && HAVE_PAYLOAD_SPACE(chan, AHRS3))
    {
        struct Location loc {};
        ekf2.getLLH(loc);
        Vector3f euler;
        ekf2.getEulerAngles(-1,euler);
        mavlink_msg_ahrs3_send(chan,
                               euler.x,
                               euler.y,
                               euler.z,
                               loc.alt*1.0e-2f,
                               loc.lat,
                               loc.lng,
                               0, 0, 0, 0);
    }
#endif
}

peterbarker, 2020-11-25:
Is this still a problem?

IamPete1, 2021-08-28:
I guess not....

closed 2021-08-28 by IamPete1