ArduPilot/ardupilot #13440: AP_Motors: add CW_X for Hexa, Octa, OctaQuad

rmackay9, 2020-01-31:
We've got the [MOTOR_FRAME_TYPE_CW_X](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_Motors_Class.h#L60) implemented for quad frames but we should also implement it for Hexacopter, Octacopters and OctaQuadcopters.

Hexacopters and Octacopters are probably pretty straight forward, for OctaQuadcopters the top motor should be before the lower motor.  I.e. top front-right motor should be 1, bottom front-right should be 2, top back-right should be 3, etc..

We could also do Dodecahexa copters although the existing "X" implementation is the same.  So we could just have the case CW_X: just below the existing case X.

rmackay9, 2020-01-31:
@rishabsingh3003 this one is quite similar to the PR you did a couple of days ago so this might be a good one for you.  No pressure though.

rishabsingh3003, 2020-01-31:
@rmackay9 I would like to work on it! Will submit a PR soon. Thanks :)

PR #13446 (2020-02-19): AP_Motors: add CW_X for Hexa, Octa, OctaQuad
This PR fixes #13440 
It adds support for CW_X frame type in Hexa, Octa, OctaQuad.
closed 2020-02-19 by rmackay9